lgsvl/simulator #1296: RoadRunner Map import error

LEE-HONGGYU, 2021-03-15:
Hello

We imported Map into RoadRunner and exported it in Unity format. 
At first, there was no problem to call it up in the normal Unity Project (ver 2019.3.3f1), but when working in LG SVL simulator (ver 2019.3.3f1), the Material was called the Shader format of Custom/RoadRunnerShader, so it did not work.

Although it is the same version, it has been well received in normal Unity as shown in the following picture, but when it is received in the lgsvl project folder, it is imported in pink like the following picture.
[image]

[image]

I'd like to hear two answers here.

1. I tried to set up the material as the link below, but I couldn't find the method. I'd appreciate it if you could tell me more detailed ways. 
https://github.com/lgsvl/simulator/issues/1074

2. I want to use the Build HD Map Mesh function through premium, is it possible if I purchase premium? How can I buy it?

LEE-HONGGYU, 2021-03-15:
In addition, is 2019.3.3f1 version the latest version of LGSVL simulator?

EricBoiseLGSVL, 2021-03-15:
@LEE-HONGGYU latest release is 06 and Unity requires 2019.3.15f1. See [here](https://github.com/lgsvl/simulator/releases/tag/2020.06).
1) This is a detailed explanation on what needs to be done to support Roadrunner meshes.  What part is causing confusion?
- convert roadrunner materials to standard unity materials.
- export map as a unity package
- import into simulator
- use unity conversion tools to convert materials to HDRP
2) Map mesh functionality will be free and will be released very soon.  We are wrapping up the release right now. :)

Taeng-ioio, 2021-03-16:
@EricBoiseLGSVL 
Hi EricBoiseLGSVL, I can't convert roadrunner materials to standard unity materials.
After Click Extract Materials, it's still pink materials(Shader is Custom/RoadRunnerShader)
How can i convert that material into Unity Standard Materials?

Taeng-ioio, 2021-03-16:
@EricBoiseLGSVL 

I finished changing Materials to Standard format. However, another problem occurred.

The following is a detailed explanation of the processes I have done.
1. Export from RoadRunner to Unity format (Select Embedded Texture)
2. download RoadRunner Import Tool into LGSVL simulator
3. Load exported files from RoadRunner to LGSVL
4. Extract Textures (png files were extracted from the model.)
5. Extract Materials (pink colored materials were created in the model.)
6. Convert the generated pink colored materials to standard format (At this time, it was confirmed that the png file obtained from texture extraction was contained in Albedo.)
7. Change Standard Materials into HDRP/Lit, (Render Pipeline - Upgrade selected Materials to High Definition Materials)
8. Drag the model into Scene to create it
9. Create folder in External/Environments, save Scene, and build

At this time, an error occurred during the build process. 
Error building Player because scripts have compile errors in the editor
The built map file showed a problem with the capacity of 0 kb.

How can you solve it? Thank you.

ps.
I tried to build an example of CubeTown, and there was the same problem.

Let's try reinstalling the Simulator.

EricBoiseLGSVL, 2021-03-17:
You need to resolve script errors before you can build anything in Unity Editor.  What are the compile errors?

closed 2021-03-25 by LEE-HONGGYU