lgsvl/simulator #1297: Question about AutonomousStuff Parking lot map

qiongwu1003, 2021-03-17:
Hi LGSVL team,

I refere this [issue ](https://github.com/lgsvl/simulator/issues/1266)update Apollo to 6.0, but when I load the AutonomousStuff Parking lot map, I can not see the parking space id in the Apollo Dreamview. 
I downlod the map [here](https://content.lgsvlsimulator.com/maps/autonomoustuff/), where there is no Apollo 6.0 HD map, I use 5.0 instead.
Is the issue caused by the map? 

Thanks in advance.

lemketron, 2021-03-17:
There is an LGSVL fork of Apollo 5.0 which is where all of the Apollo maps live, but for Apollo 6.0 and master all of the LGSVL changes were upstreamed to the main Apollo repository so there is no need for an LGSVL fork of 6.0 or master.

The apollo 5.0 maps should work for 6.0 as well if you put them in the correct directory. Be sure to get the latest maps from the (LGSVL fork of the) Apollo 5.0 repository.

qiongwu1003, 2021-03-17:
@lemketron Thanks for your reply.
I do use the map you mentioned. But I can not see the parking space id like below(both Apollo master and r6.0.0).
[image]

lemketron, 2021-03-17:
@qiongwu1003 I think you might find this thread (and comment) helpful:
https://github.com/lgsvl/simulator/issues/1266#issuecomment-789546597

qiongwu1003, 2021-03-17:
@lemketron 
OK, thank you very much. I will try other ways.

closed 2021-03-17 by qiongwu1003