lgsvl/simulator #1298: How to make NPC accelerate smoothly between waypoints?

lzq0115, 2021-03-19:
Hello LGSVL team,

I set some waypoints with different speed for a NPC car with Python-API. I expect it will interpolate the speed difference when it drives between two waypoints. In fact, it will drive with a constant speed set by waypoint #1 and jump to the speed set by waypoint #2 suddenly when it arrives at waypoint #2. How can I deal with it without building a new simulator?

Thanks.

EricBoiseLGSVL, 2021-03-19:
This is a request that we plan on fixing soon.  There is no way to change this without changing source code.

lzq0115, 2021-03-20:
> This is a request that we plan on fixing soon. There is no way to change this without changing source code.

For now I try to fix it in a stupid way. That is dividing the acceleration process into more waypoints. Also, thanks for the hard work. Hope the new function will be online soon.

EricBoiseLGSVL, 2021-03-22:
ok sounds like a reasonable way to get what you need.  Waypoint refactor won't be out this release but soon after.  We plan on more regular releases moving forward.