lgsvl/simulator #1299: How to make custom map from LGSVL example map

kjs675252, 2021-03-19:
Hi! I'm trying to change speed limit of map lane via 'Unity' to limit Asset vehicle's velocity.

First, I succeed 'build process' from lgsvl_manual without any AssetBundles such as Vehicle, Environment and Sensors.

And I download 'Cubetown' from "https://github.com/lgsvl/CubeTown" to simulator repository and then I 'only' change speed limit of MapLane01, MapLane02 via Unity.

Finally I build the simulator and there is no error while build process.

But at the 'AssetBundles' repository, the 'environment_CubeTown' file empty(0 byte). I think it is strange result.

Anyone can help me? Thank you for your help.

lemketron, 2021-03-19:
Did you review the information at https://www.lgsvlsimulator.com/docs/assets/ ?

kjs675252, 2021-03-19:
@lemketron 

Thaks for your advice!

I follow the manual and then get some error message

Do I have to solve all of them?

========================================================================
Checking...
ERROR: File 'Rain.vfx' with '.vfx' extension is not allowed inside '/Assets/Effects'
ERROR: File 'Snow.vfx' with '.vfx' extension is not allowed inside '/Assets/Effects'
ERROR: Folder 'HDRPDefaultResources' should not be inside of '/Assets'
ERROR: File 'ExampleFMU.fmu' with '.fmu' extension is not allowed inside '/Assets/Resources'
ERROR: File 'VFXDefaultResources.asset' should not be inside '/Assets'
ERROR: File '/Assets/Scripts/Dynamics/FMU/FMU.cs' does not have correct copyright header
ERROR: File '/Assets/Scripts/Editor/FMUEditor/FMUImporter.cs' does not have correct copyright header
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/Materials/EnvironmentMaterials/EnvironmentDamageAlbedo.png'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/Materials/EnvironmentMaterials/EnvironmentDamageHeight.png'
ERROR: Asset '/Assets/External/Environments/CubeTown/CubeTown.unity' depends on '/Assets/Materials/EnvironmentMaterials/EnvironmentDamageNormal.png'
ERROR: File name 'Foliage Diffusion Profile.asset' contains spaces in '/Assets/HDRPDefaultResources'
ERROR: File name 'Foliage Diffusion Profile.asset.meta' contains spaces in '/Assets/HDRPDefaultResources'
ERROR: File name 'Scene PostProcess Profile.asset' contains spaces in '/Assets/HDRPDefaultResources'
ERROR: File name 'Scene PostProcess Profile.asset.meta' contains spaces in '/Assets/HDRPDefaultResources'
ERROR: File name 'Skin Diffusion Profile.asset' contains spaces in '/Assets/HDRPDefaultResources'
ERROR: File name 'Skin Diffusion Profile.asset.meta' contains spaces in '/Assets/HDRPDefaultResources'
ERROR: File name 'Sky and Fog Settings Profile.asset' contains spaces in '/Assets/HDRPDefaultResources'
ERROR: File name 'Sky and Fog Settings Profile.asset.meta' contains spaces in '/Assets/HDRPDefaultResources'
Done!
========================================================================

EricBoiseLGSVL, 2021-03-19:
No check is more for internal development.  That is not an indicator of issues.  Did you make any other changes to the map?  Did you make sure to have both Windows/Linux build modules installed when you installed Unity Editor?

kjs675252, 2021-03-22:
@EricBoiseLGSVL 

Thanks for your comment! 

As you said, I didn't install support for Linux and I try to solve it.

Because I use Window OS for my simulation environment I try to install support for Linux.

But how can I install UnityHub.Appimage in Window OS?

Should I use WSL(Windows Sybsystem for Linux)?

EricBoiseLGSVL, 2021-03-22:
no in unity hub, you can install support for windows.  It's in the documentation [here](https://www.lgsvlsimulator.com/docs/build-instructions/) 2019.3.3.15f1 and a git hub issue here #1258

kjs675252, 2021-03-23:
@EricBoiseLGSVL 

Thanks a lot for your comment!!

I install version of 2019.3.15f1 and finally build my own custom map.

But there is one weird thing.

I download 'CubeTown' map from lgsvl github, and open it at Unity 2019.3.15f1.

However I can't see any road lane even though it is original file.

Could you help me about this problem?

[image]

EricBoiseLGSVL, 2021-03-24:
It looks like you are missing the textures.  Did you clone the repo and build this map or did you download a zip?  You need to enable git lfs and clone the repo.