lgsvl/simulator #1300: Is there any automatic way to do map annotation?

Addi2020, 2021-03-20:
map annotation by manual method in unity software is taking too much time. So is there any way to do map annotation automatically?

EricBoiseLGSVL, 2021-03-22:
If you have an HD map then importing is relatively automatic. Do you have a digital twin environment?

Addi2020, 2021-03-26:
@EricBoiseLGSVL digital twin environment means what?
I have point cloud map (,pcd file), I import this file the do map annotation to create HD map. Adding lanes, U turns, boundaries, Traffic lights, It take too much time for adjustment especially Traffic lights for position adjustment for detection of light. 
So is there any faster and automatic way to create HD map by importing .pcd file? or create HD map directly by using other technique?

EricBoiseLGSVL, 2021-03-31:
There is no automatic way to create a hdmap from point cloud data, sorry.  There are a few companies that offer this as a service, but it is not free.  After you create the hdmap, why are you needing to adjust it?  You should align the hd map elements with the point cloud and then create the map.

Addi2020, 2021-03-31:
Got it.
As the point cloud data is not very clear and also not correct like real world , So we need to align or adjust the lanes after testing. especially the adjustment problem is for traffic lights, for 3d .pcd file the height and the distance between two lights are not the same as real world. it is very difficult to adjust traffic lights. Also one pole contains two traffic lights so if the distance between lights is not right then it overlaps with each other and the traffic light detection is not possible.

EricBoiseLGSVL, 2021-03-31:
Cool, yes this is a common issue with point clouds.  We are looking into ways to make this easier currently.  We hope to release more info on our progress soon.

lemketron, 2021-07-06:
If you are able to obtain an existing HD map (for example if this is a real world location and you are also doing testing in a real vehicle, you probably also have an HD map in OpenDrive or Lanelet2 format) then you could import that HD map into the simulator environment so you don't have to create all of the annotations manually.

As @EricBoiseLGSVL mentioned, we are looking into ways to make this easier.

closed 2021-07-06 by lemketron