lgsvl/simulator #1302: KeyError: 'transform'

lzq0115, 2021-03-23:
Hello, lgsvl team.
I am using PythonAPI and gym-lgsvl to control lgsvl and I am using it to perform a reinforcement learning which needs to interact with the environment frequently.(I guess this would be the problem?) I read issue #990, he got this error because of not using sim.run(). But I use sim.run(time_limit=1/20, time_scale=10) function.  Mostly the code works well and is able to finish the whole job. But when this error appears, all I can do is stop it and start it over, which is frustrating. I get this error randomly and I cannot reproduce this error. I can get this error both on Windows 10 and Ubuntu 16.04 with lgsvl 2020.06 version. 

Here is a picture of the variable that cause the error and error info is as follows.
[image]

Traceback (most recent call last):
  File "/home/trevor/Downloads/pycharm-2020.1/plugins/python/helpers/pydev/pydevd.py", line 1438, in _exec
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "/home/trevor/Downloads/pycharm-2020.1/plugins/python/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/home/trevor/lgsvl_RL/main.py", line 78, in <module>
    s_prime, reward, done, info = env.step(v_action)
  File "/home/trevor/anaconda3/envs/lgsvl/lib/python3.7/site-packages/gym/wrappers/time_limit.py", line 16, in step
    observation, reward, done, info = self.env.step(action)
  File "/home/trevor/lgsvl/gym-lgsvl-master/gym_lgsvl/envs/lgsvl_env_new.py", line 123, in step
    print("real speed is " + str(self.ego.state.speed))
  File "/home/trevor/lgsvl/PythonAPI-master/lgsvl/agent.py", line 114, in state
    return AgentState.from_json(j)
  File "/home/trevor/lgsvl/PythonAPI-master/lgsvl/utils.py", line 60, in from_json
    Transform.from_json(j["transform"]),
KeyError: 'transform'

EricBoiseLGSVL, 2021-03-24:
Are you in interactive mode?

lzq0115, 2021-03-25:
> Are you in interactive mode?

Nope, interactive mode is not chosen.

EricBoiseLGSVL, 2021-03-25:
Sorry we will need more info to debug this.  Can you post the api script you are running or a simplified one that we can see the error?  @hadiTab any idea?

lzq0115, 2021-03-26:
I think only the whole project would find the problem, so I post it on my github. https://github.com/lzq0115/lgsvl_RL.git
Please have a look. The code may be poorly written. Hope this will not be a trouble for you.

EricBoiseLGSVL, 2021-03-30:
No we cannot troubleshoot the entire project.  Please post the api script you are running here and we can see if we can reproduce the issue.

lzq0115, 2021-04-02:
Sorry for the late reply. Here is the api script.
```python
import gym
from gym import error, spaces, utils
from gym.utils import seeding
import numpy as np
import lgsvl
from lgsvl.geometry import Vector, Transform
import os
import random
import math
import cv2

CONFIG = {
    "scene": "WideFlatMap",
    "port": 8181,

    "action_space":
        spaces.Box(
            np.array([-1], dtype=np.float32),
            np.array([+1], dtype=np.float32)
        ),  # throttle (+) / brake (-)

    "observation_space":
        spaces.Box(
            low=np.array([-10, -25], dtype=np.float32),
            high=np.array([50, 25], dtype=np.float32)
        )  # delta_dis & delta_v
}
step_size = 1/20
initial_speed = 10
max_step = 900

class LgsvlEnvNew(gym.Env):
    metadata = {'render.modes': ['human']}

    def __init__(self, config=CONFIG):

        self.env = lgsvl.Simulator(os.environ.get("SIMULATOR_HOST", "127.0.0.1"), config["port"])
        if self.env.current_scene == config["scene"]:
            self.env.reset()
        else:
            self.env.load(config["scene"])

        self.spawns = self.env.get_spawn()
        self.vehicles = dict()
        self._occupied = list()
        self.ego = None
        self.npc = None
        self.seed()
        self.control = lgsvl.VehicleControl()
        self.forward = None
        self.origin = None

        self.action_space = config["action_space"]
        self.observation_space = config["observation_space"]
        self.v = 0
        self.v_old = 0
        self.delta_dis = 0
        self.delta_v = 0
        self.accel = 0
        self.desired_speed = 10
        self.prev_cte = 0.0
        self.int_cte = 0.0

        self.count = 0

        self.reward = 0
        self.done = False

[... 260 lines cut ...]
              lgsvl.DriveWaypoint(point[35], 6, self.ego.transform.rotation),
              lgsvl.DriveWaypoint(point[36], 5, self.ego.transform.rotation),
              lgsvl.DriveWaypoint(point[37], 4, self.ego.transform.rotation),
              lgsvl.DriveWaypoint(point[38], 3, self.ego.transform.rotation),
              lgsvl.DriveWaypoint(point[39], 2, self.ego.transform.rotation),
              lgsvl.DriveWaypoint(point[40], 1, self.ego.transform.rotation),
              lgsvl.DriveWaypoint(point[41], 0, self.ego.transform.rotation)]
        for i in range(42):  # set to 42
            if self.npc.state.position.z < point[i].z:
                waypoints.append(wp[i])
        if waypoints:
            self.npc.follow(waypoints)

    def set_ego_direction(self):
        s = self.ego.state
        s.transform.rotation = self.forward
        self.ego.state = s

    def update_origin(self):  # no need, but better than no
        self.origin[0].y = self.ego.state.transform.position.y

    def too_short_punish(self):
        if self.count < max_step:
            self.reward -= 20000 * (max_step - self.count)
```

EricBoiseLGSVL, 2021-04-02:
Thanks for the file.  I'll need to ask @daviduhm and @hadiTab on the open ai gym integration.  It seems odd that it fails to return the transform.  Have you tried the latest simulator and python api?

lzq0115, 2021-04-03:
Not yet, I don't even know the new version was released. I will try it.

lzq0115, 2021-04-03:
OK, I try with the new version 2021.1 and new python API. The problem will appear even much more earlier than the old version 2020.6.