lgsvl/simulator #1303: How to get the ego car reversed through lgsvl_msgs/msg/VehicleControlData

ZhenshengLee, 2021-03-24:
Hi,

I am tring to get the latest autoware.auto 1.0.0 avp demo work in my ubuntu1804 environment, and I found the issue when testing the simulator
```
ros2 topic pub --once /lgsvl/vehicle_control_cmd lgsvl_msgs/msg/VehicleControlData "{
> header:
>   {stamp:
>     {sec: 1600775035,
>     nanosec: 496432027},
>   frame_id: map},
> acceleration_pct: 1.0,
> braking_pct: 0.0,
> target_gear: 2
> }
> "
publisher: beginning loop
publishing #1: lgsvl_msgs.msg.VehicleControlData(header=std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=1600775035, nanosec=496432027), frame_id='map'), acceleration_pct=1.0, braking_pct=0.0, target_wheel_angle=0.0, target_wheel_angular_rate=0.0, target_gear=2)
```
the ego car cannnot get reversed, and it get forward, the same as `target_gear: 0`
So, what's the right msg to let the ego car get reversed? 

thanks a lot!

EricBoiseLGSVL, 2021-03-24:
@ZhenshengLee Sorry this is not supported at this time.  We only have reverse for apollo control.  You will need to implement this.  Right @hadiTab ?  For 06 release you would need to change the ROS2 conversion data and then edit the controller to set the vehicle Reverse == true from message received.  Look at Reverse code for CyberRT and ApolloControl in VehicleControlSensor.cs

```
if (data.CurrentGear == GearPosition.Reverse)
{
    Dynamics.ShiftReverseAutoGearBox();
}
else if (data.CurrentGear == GearPosition.Drive)
{
    Dynamics.ShiftFirstGear();
}
```

EricBoiseLGSVL, 2021-03-25:
I have heard from @lemketron that this is supported in our new release out very soon :)

ZhenshengLee, 2021-03-26:
Thanks a lot!
@EricBoiseLGSVL You mean in the next release, the ros2 lgsvl msgs, including /vehicle_control_cmd, /vehicle_state_cmd, /gnss_odom, /state_report, will get expeteced behavior, right?

ZhenshengLee, 2021-03-26:
@EricBoiseLGSVL thanks a lot
But I found nothing happend in 202101 release to get the ego car reversed

ZhenshengLee, 2021-03-26:
@EricBoiseLGSVL  Can you briefly describle the feature implementation? Will the ego car be reversed control throught state_cmd or control_cmd?

TrentWeiss, 2021-03-27:
Can confirm: using "target_gear: 2" (reverse) that is part of lgsvl_msgs/msg/VehicleControlData _does not_ make the vehicle reverse, even in the new release.

ZhenshengLee, 2021-03-29:
Badluck, In 2021.01 turning control is not work too.
```sh
ros2 topic pub -r 10 /lgsvl/vehicle_control_cmd lgsvl_msgs/msg/VehicleControlData "{
header:
  {stamp:
    {sec: 1600775035,
    nanosec: 496432027},
  frame_id: map},
acceleration_pct: 0.0,
braking_pct: 0.0,
target_wheel_angle: 2.0,
target_wheel_angular_rate: 1,
target_gear: 0
}
"
```

ZhenshengLee, 2021-03-29:
> Badluck, In 2021.01 turning control is not work too.
> 
> ```shell
> ros2 topic pub -r 10 /lgsvl/vehicle_control_cmd lgsvl_msgs/msg/VehicleControlData "{
> header:
>   {stamp:
>     {sec: 1600775035,
>     nanosec: 496432027},
>   frame_id: map},
> acceleration_pct: 0.0,
> braking_pct: 0.0,
> target_wheel_angle: 2.0,
> target_wheel_angular_rate: 1,
> target_gear: 0
> }
> "
> ```
due to autoware auto and lgsvl interface incompatibility, 
you can comment the line in `Assets/Scripts/Bridge/Ros2/Ros2Conversions.cs`  to solve this.
`SteerInput = data.target_wheel_angular_rate, `

TrentWeiss, 2021-03-29:
> Badluck, In 2021.01 turning control is not work too.
>
> ```shell
> ros2 topic pub -r 10 /lgsvl/vehicle_control_cmd lgsvl_msgs/msg/VehicleControlData "{
> header:
>   {stamp:
>     {sec: 1600775035, [...]

You have set target_gear to 0, which is neutral. Hence the car not moving.  You need to set target_gear to 1 (drive).  Also, I am able to steer the car by only setting target_wheel angle and leaving target_wheel_angular_rate at 0.0.

EricBoiseLGSVL, 2021-03-29:
Sorry for the confusion.  Reverse is not supported in the latest release.  A team created their own control sensor that supports this.  You can look at Apollo control sensor and apollo data conversion classes to see how to implement.

EricBoiseLGSVL, 2021-03-29:
adding @hadiTab so we can add fixes to next release

marcusvinicius178, 2021-08-31:
Hi @EricBoiseLGSVL a short question, please. The issue of bridge connection Autoware.Auto -SVL new 2021 release discussed here https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/issues/1264#

Was fixed just for 2021.1 release or the connection works for all other 2021 releases(2021.1.1, 2021.1.1, 2021.2.1, 2021.2.1, etc...)?????

I wish to know to use the correct SVL version for Autoware.Auto

Thanks ☺️

EricBoiseLGSVL, 2021-09-01:
Yes, we are working on this now to validate autoware.auto.  We should have a solution soon.

closed 2022-04-07 by ZhenshengLee