lgsvl/simulator #1305: How to

ideasplus, 2021-03-25:


EricBoiseLGSVL, 2021-03-30:
@ideasplus You can run the simulator in Editor and set debug logs when the agent is added or create a sensor plugin that will output this data to the visualizer.

EricBoiseLGSVL, 2021-03-31:
No problem, glad to help.

What simulator version are you using?
06 release MUST be Unity 2019.3.15f1
2021.1 release MUST be Unity 2019.4.18f1
Latest documentation is for 2021.1

As for Unity help:
First way: As for the code you will need to add Debug.Log(Data you want); So in AddAgent, add the code to calculate what data you want and then Debug.Log to Unity Console Panel.  You will need to make a new binary of simulator.

Second way: Create a new sensor (Copy a simple one like KeyboardControlSensor), DO NOT use the same name for the class, Add new functionality to get all NPCs in scene and on a keypress display data you want in the visualizer method or again with debug.  You can create any functionality you want with this sensor.  Just be sure to add the sensor to the vehicle sensor config.  Again, see the keyboard sensor as an example.

If you need custom data/behavior with simulator, be sure to watch some tutorial videos on Unity.  This will help immensely when first altering simulator.