lgsvl/simulator #1308: Sensor config editor uses left-handed coordinate systems

TrentWeiss, 2021-03-26:
I have noticed that in the new pre-release tag (version 2021.1), the browser-based editor for creating/modifying Sensor configurations appears to use a convention of left-handed coordinate systems. Take a look at the screenshot below.  X cross Y (the mathematical definition of the Z vector for a Euclidean space) should be pointing to the right, but it is pointing left in the editor, indicating the editor uses a left-handed convention.  You could make the argument that this is just a choice of convention and consistency is all that matters and be (mostly) right, but the overwhelming majority of the robotics and automation world uses a right-handed convention.  This will present a sign-error nightmare when trying to convert these LGSVL sensor config files to/from other geometric definition systems like ROS's URDF.

[image]

TrentWeiss, 2021-03-26:
I thought this might be my mind playing tricks on me with isometric views, but rotating the coordinate system shows the same left-handed pattern.
[image]

zelenkovsky, 2021-03-28:
We are going to change it to ROS coordinate system in our next website update.

TrentWeiss, 2021-03-29:
> We are going to change it to ROS coordinate system in our next website update.

Excellent.

Is the source code for the website open-source anywhere?

TrentWeiss, 2021-04-14:
Any update on this?  The left-handed notation in the web-based GUI is rendering it effectively useless for us (all of our existing codebase assumed a right-handed notation).  We also don't want to waste our time creating sensor configurations under a convention that will be eminently changing.

@lemketron

zelenkovsky, 2021-05-05:
We are working on this change, you can safely create your sensor configurations right, it will be automatically translated.

The cloud part is not open source, private cloud deployment for your company or project is part of a commercial product. Please request a demo at https://www.svlsimulator.com/ to get more details.