lgsvl/simulator #1309: Can't run API Only Template in SVL 2021.1

vatanaksoytezer, 2021-03-26:
Hi, I was using 2020.06 with API Only configurations for a while. When I saw 2021.1 is release I immediately downloaded and started to transfer my work to this version as there are incredible new features on testing that I am willing to use.

However although I created an API Only simulation, when I click run simulation nothing happens. And I see nothing being changed in the UI in my local computer. I am using Linux binary of 2021.1 SVL simulator on Ubuntu 18.04. There seems to o problem on my machine connection to cloud as I can verify that my machine can be seen online in clusters tab.

An extra unrelated issue is that it takes a lot of time to connect to cloud and also usually simulations come up very slow in the related tab.

lokesku, 2021-03-27:
How did you create an API only simulation ? Please follow instructions here: https://www.svlsimulator.com/docs/running-simulations/running-simulator/#apionlysimulation

Please make sure your newly created simulation shows "Idle" before you click "Run simulation". Once you click run, your local simulator should show "API ready!". Now you can run any of the python scripts to load assets and scenario into your simulation: https://www.svlsimulator.com/docs/python-api/python-api/

If problem still persists, please share your Player.log (~/.config/unity3d/LGElectronics/SVLSimulator/Player.log)

lokesku, 2021-03-27:
>>An extra unrelated issue is that it takes a lot of time to connect to cloud and also usually simulations come up very slow in the related tab.
Please make sure you are using a decent speed internet connection. Simulator needs to download your assets also from cloud, which could be in GBs depending on map you choose.

Also make sure your firewall is not blocking the connection.

vatanaksoytezer, 2021-03-27:
> How did you create an API only simulation ? Please follow instructions here: https://www.svlsimulator.com/docs/running-simulations/running-simulator/#apionlysimulation

I did as the instructions suggested. And I can correctly identify that simulation shows "Idle". I think my internet connection is pretty stable, and since I am using API Only I think it shouldn't download anything before I start the script, am I wrong?

For the firewall issues, I don't use firewall at all and had no problems on any other site. I will give it a few more tries and send the log soon.

vatanaksoytezer, 2021-03-27:
Well, I honestly don't know the issue back then but it works as expected right now. I am sending the previous log in case you may find something useful in it.
[sendable.log](https://github.com/lgsvl/simulator/files/6216115/sendable.log)

closed 2021-03-27 by vatanaksoytezer

hadiTab, 2021-03-27:
Just wanted to point out that unlike the previous release where assets were
downloaded immediu after adding them in the Web UI, in the new release
assets added to your library are not download until their first use. So the
first API Only simulation run would first download the map and ego vehicle
before starting. Subsequent runs wouldn't take as long as the assets will
be cached locally.