lgsvl/simulator #1310: Error when signing up a new SVL cloud account

AIasd, 2021-03-26:
[image]

Hi team,

I am encountering error when trying to sign up an account. Any ideas on this error?

Thanks!

TrentWeiss, 2021-03-26:
I am also noticing the new cloud-based system is quite buggy (although I was eventually able to create an account).

Most of the features spelled out in the docs result in the system just hanging for a few minutes, possibly followed by a largely unhelpful error message like "'asdf' was not created" (where 'asdf' is the name of the new Sensor config I was trying to make).

Perhaps their cloud-based backend is overwhelmed by people excited to use the new release?

AIasd, 2021-03-26:
After trying again several times at different time, I succeeded creating the account finally.

closed 2021-03-26 by AIasd

lokesku, 2021-03-27:
Thanks for your report. We are continuously improving and with cloud based web user interface, we will be able to roll out updates with fixes more often now.