lgsvl/simulator #1312: Indianapolis Motor Speedway map GPS coordinates are not correct

mslavescu, 2021-03-28:
The orientation and GPS origin of the map seems to be off, compared with iRacing track right side, that can be downloaded from Z1 Dashboard (part of the txt file also).

Indianapolis Motor Speedway map
https://wise.svlsimulator.com/maps/profile/62765742-57bf-4ccd-85e5-db8295d34ead

Here you can see some of my calculations trying to reposition a GPS trace captured by doing a few laps on the new LGSVL 2021.1 simulator.

[indianapolis-iRacing-ovalTrackDataRight-vs-LGSVL-driving-keyboard.xlsx](https://github.com/lgsvl/simulator/files/6218307/indianapolis-iRacing-ovalTrackDataRight-vs-LGSVL-driving-keyboard.xlsx)

Here is the GPS trace from the driving around, you can see how off is the LGSVL trace (red line) vs iRacing (blue):

[0328163254-74178.txt](https://github.com/lgsvl/simulator/files/6218304/0328163254-74178.txt)

[image]

Best position I could get:

[0328163806-74178.txt](https://github.com/lgsvl/simulator/files/6218309/0328163806-74178.txt)

[image]

You can load the TXT files here to analyze:
https://www.gpsvisualizer.com/draw/?api=g&bg=h&z=20&c=39.799438,-86.230606&marker=

EricBoiseLGSVL, 2021-03-30:
@mslavescu This seems like a bug with the position of the map origin.  We will look into this

mslavescu, 2021-06-25:
@EricBoiseLGSVL do you have any update on this issue?

EricBoiseLGSVL, 2021-06-25:
It is fixed and will be in the next release.  Thanks for your patience and letting us know.  It is still a bit off but since the map is not an exact digital twin, it is more precise.

mslavescu, 2021-07-05:
@EricBoiseLGSVL I extracted the coordinates from Lanet2 file from SVL IMS map:
[IndianapolisMotorSpeedway-gps.txt](https://github.com/lgsvl/simulator/files/6761493/IndianapolisMotorSpeedway-gps.txt)

Can be loaded here and viewed, just set max 1500 point before click Import:

https://www.gpsvisualizer.com/draw/?api=g&bg=h&z=20&c=39.799438,-86.230606&marker=

This should help with tuning of the algorithm for the SVL map, which is not accurate, as you can see here:

[image]
[image]

Would be possible to update the IMS map to match as close as possible the actual left/right sides of the track?

EricBoiseLGSVL, 2021-07-06:
In our next release we have a tool that can try to reduce errors with map origin.  This is the best we can do since this map is not a digital twin but an approximation.  We don't have plans currently to redo this map sorry.

lemketron, 2021-07-06:
I'd be curious to hear more about why the exactness of the track matters. Are you trying to do some training that is location-specific? 

We do not expect (or support) the exported HD maps from this environment to be used in any real world in-vehicle testing, so hopefully that's not it...

closed 2021-07-16 by EricBoiseLGSVL