lgsvl/simulator #1315: A machine cannot belong to two or more clusters at once

MisterioRemo, 2021-03-29:
In the 2021.1 version of the simulator, a machine cannot belong to two or more clusters at once.
For example, I have a cluster of two machines, everything works.
Now I want to work from one machine (localhost).
I need to create a cluster of one machine, this is not possible as long as the machine is part of another cluster. So, it is necessary to delete the previous cluster, unlink the simulator, then link to the cloud again and finally create a new cluster. 
It's a bit tiresome.

EricBoiseLGSVL, 2021-03-30:
@MisterioRemo Yes, this is by design.  @zelenkovsky are there any plans change this?

MisterioRemo, 2021-03-30:
@EricBoiseLGSVL thank you for the answer. Since this was possible in the previous version, I thought it was a bug.

roshambo919, 2021-10-05:
I am also curious about this. Similarly, I no longer see the ability to add a machine with just an IP address. Instead, I have to download the binary and run `./simulator`, but will this work linking a machine where I only have terminal (ssh) access to it?