lgsvl/simulator #1318: SVLSimulator can not receive ControlCommand

zhp-ncu-dev, 2021-03-29:
Hi,
I use SVLSimulator-2021-1 with Apollo3.0, but SVLSimulator could not receive pb_msgs/ControlCommand. 

thanks!

EricBoiseLGSVL, 2021-03-30:
@lemketron is Apollo 3.0 even supported in 2021.1?

zhp-ncu-dev, 2021-03-30:
> @lemketron is Apollo 3.0 even supported in 2021.1?

zhp-ncu-dev, 2021-03-31:
> @zhp-ncu-dev can you provide more details about how you setup the simulation?
> To create an Apollo 3.0 simulation you will need to create a sensor configuration that uses the `ROS Apollo` bridge. The easiest way to do this would be to "clone" the `Apollo 5.0` sensor configuration for the `Lincoln2017MKZ` vehicle and just change the bridge type to `ROS Apollo`. If you select this sensor configuration for a simulation you will then be given the option to choose `Apollo 3.0` as the autopilot.

i have solved this problem, and thank you very much.

closed 2021-03-31 by zhp-ncu-dev

marcusvinicius178, 2021-06-09:
Hi @zhp-ncu-dev @hadiTab I still could not run Apollo 3.0 with LGSVL simulator. I have repeated the steps above to set the bridge as you can check here: https://youtu.be/CWPqeUdyQPI

But I got this error:

[image]

Could you please provide me which is the correct repository of Apollo 3.0 that I must clone? Can send me the link?
After this provide me the proper branch I must checkout? And finally if I need to build with special flag etc?
I am struggling for days to build Apollo3.0 correct if I get simulator version there is not the rosbridge.sh file. Then the build will not work: https://github.com/lgsvl/simulator/issues/1475

Thanks in advance