lgsvl/simulator #1320: rain is not visible with python api control

20171CSE0680, 2021-03-30:
I am trying to simulate all weather conditions but rain is not working with API but with normal simulation, it is changeable and visible.

i am using version 2021.1 (because version 2020.6 is not connecting to python API
[image]
[image]
[image]

)

mslavescu, 2021-03-30:
For 2020.06 use the corresponding tag from PythonAPI, master works only with 2021.1:
https://github.com/lgsvl/PythonAPI/tree/2020.06

EricBoiseLGSVL, 2021-03-30:
@20171CSE0680 we will look into the API weather control with API, thanks for the heads up.  @mslavescu Thanks for helping with 06 API issue.

rajeshjoshi8792, 2021-06-25:
i m getting same problem. rain is not working using api.

EricBoiseLGSVL, 2021-06-28:
fixed and will be in next release.  Thank you for posting this issue.  Rain changed to use the visual effect graph particle system and needed a change to the way it is initialized.

closed 2021-06-28 by EricBoiseLGSVL