lgsvl/simulator #1322: Visual Scenario Editor with Apollo3.0.

zhp-ncu-dev, 2021-04-01:
Hello:
I have implemented the communication among Apollo3.0 , Python API and Dreamview API, but could not run a simulation with a json scenario created by VSE. 

thanks.

zhp-ncu-dev, 2021-04-01:
I have send the map to Apollo3.0 Dreamview , but could not send the destinationPoint of json file to Dreamview.

EricBoiseLGSVL, 2021-04-01:
Apollo 3.0 is not supported with the VSE runner sorry.

zhp-ncu-dev, 2021-04-02:
> Apollo 3.0 is not supported with the VSE runner sorry.
thanks. But I want to know the communication between Apollo5.0 and VSL, and I maybe solve this problem. Can you give me a hand? Thank you very much!

zhp-ncu-dev, 2021-04-02:
thanks. But I want to know the communication between Apollo5.0 and VSL, and I maybe solve this problem. Can you give me a hand? Thank you very much!

EricBoiseLGSVL, 2021-04-02:
The api's for 3.0 might not be available.  I'll need @hadiTab or @lemketron to confirm

lemketron, 2021-04-14:
Can you explain exactly what you're trying to do? It sounds like you have created and saved a scenario with VSE, and you are trying to run that scenario with Apollo 5.0 (or previously Apollo 3.0)? 

To do this you would need to use the VSE runtime template as documented here:
https://www.svlsimulator.com/docs/running-simulations/running-simulator/#vsesimulation

As noted in the documentation, VSE is only supported with Apollo 5.0 (certainly not older Apollo 3.x, though newer Apollo like 6.0 may also work).

Are you trying to do something with Python API and Dreamview API as well? You cannot use the saved VSE scenario JSON file with Python API but must run it with the VSE runtime template as described in the above documentation link.

marcusvinicius178, 2021-06-04:
Hi @EricBoiseLGSVL , @lemketron  Is this tutorial wrong: https://www.svlsimulator.com/docs/archive/2020.06/apollo-instructions/ ?

Please take a look on this issue: https://github.com/lgsvl/simulator/issues/1475

EricBoiseLGSVL, 2021-06-05:
Are you running the 2020.06 release with this tutorial?

marcusvinicius178, 2021-06-05:
The issue is that the bridge bridge does not run:
./scripts/rosbridge.sh
Returns errors that the websocket pkg is not found. I believe the repo to be cloned in this tutorial is wrong. Or at least it should inform which release, branch must be checkout to get the ros_pkgs and Ros bridge that are not present in the repo from tutorial.

EricBoiseLGSVL, 2021-06-05:
Why are you on 06 release working with apollo 3.0?  This is a fairly old tutorial.

marcusvinicius178, 2021-06-05:
> Why are you on 06 release working with apollo 3.0? This is a fairly old tutorial.

Because SVL provides PYTHON API to work alongside the Bridge, therefore I can define the AGENTS of the Environment with Python and afterwards I can use the Apollo Bridge to define EGO vehicle planners. In LGSVL is one (BRIDGE) Or another (API ONLY) . I do not have option to use both simulateneously right?
In addition I am using Apollo 3.0 because a Chinese guy helped me to switch the motion planner just on this version. The other newer versions it is not feasible change the planners (almost impossible due to several scenarios, tasks interpretation for planning module). He told me 3.0 is much more robust and ready to switch planners. That is why I am using 3.0

EricBoiseLGSVL, 2021-06-05:
Ah I see.  Yes you should be able to use python and apollo.  You set the bridge in the python script.
Yes, the planners have become much more complex since 3.0.
"Not along LG SVL"?  what do mean?

marcusvinicius178, 2021-06-05:
Yes you got the point. What I mean is maybe better to explain visually:

For example in LGSVL we can have integration with Python and simulator just running Python Scripts with a chosen code editor. After run the python scripts the simulator opens with the Agents that you defined there. However the EGO vehicle will move in this scenario, following the commands we sent to the EGO vehicle in Python script (move x, turn theta degrees, etc). 
[image]

This is not what I wish!
I wish to define the Agents and scenario with Python. But I wish that the EGO vehicle get the route and the commands to control the car through the use of Dreamview (in case of Apollo) or through the use of Launch files (in case of Autoware). And this can be achieved just using SVL, as far as I know. That provides a specific area to put the python code, but let the EGO vehicle be controlled by the Launch files (Runtime Manager in Autoware.AI) or Dreamview.

[image]
[image]

EricBoiseLGSVL, 2021-06-07:
Why do you need the pythonapi interface?  ApiOnly can do the same thing.  Set the autopilot host and port.  DV. apis can be accessed there unless I am missing something @hadiTab @lemketron

marcusvinicius178, 2021-06-07:
Hummm I was imagining that the PythonApi didn't let me control the EGO VEHICLE through the AD stacks Bridge (autopilot). I thought it blocked the interface with the bridges due the port connection be already busy (9090)  by API Only application, therefore when the bridge tries to communicate using this port, the connection would be refused. 
My idea was that the PythonApi Interface had implementation was a work around improvement on this sense that allowed both python and bridges communicates with the simulator. But now after your reply I am not sure of the difference functionalities off API ONLY VS PYTHON API ...

EricBoiseLGSVL, 2021-06-07:
PythonAPI is bugged at the moment.  I would use ApiOnly.  @lemketron  @hadiTab

marcusvinicius178, 2021-06-07:
Ok good to know. Thanks!