lgsvl/simulator #1323: SVL API  sim.run()

cxf99, 2021-04-01:
sim.run () unable to send message to run again after running

What should I do to continue 

  sim.remote.command("vehicle/apply_control", self.args)

lemketron, 2021-04-02:
Sim.run() is a blocking call (if a timeout is not passed) unless a callback is triggered while the simulation is running.

You can read about it (and see an execution flow diagram) here: 
https://www.svlsimulator.com/docs/python-api/python-api/#simulation

cxf99, 2021-04-02:
> Sim.run（）是阻塞调用（如果未传递超时），除非在模拟运行时触发了回调。
> 
> 您可以在此处阅读有关内容（并查看执行流程图）：[https](https://www.svlsimulator.com/docs/python-api/python-api/#simulation) :
> [//www.svlsimulator.com/docs/python-api/python-api/#simulation](https://www.svlsimulator.com/docs/python-api/python-api/#simulation)

lemketron, 2021-04-02:
Check out quickstart script 33 for an example of how to do stepped simulation:
https://github.com/lgsvl/PythonAPI/blob/master/quickstart/33-ego-drive-stepped.py

If using Apollo then you'll want to use Apollo 6.0 or master and include the clock sensor so Apollo runs with simulation clock as noted in the above script.
https://www.svlsimulator.com/docs/simulation-content/sensors-list/#clock

cxf99, 2021-04-02:
> 请查看快速入门脚本33，以获取有关如何进行逐步仿真的示例：[https](https://github.com/lgsvl/PythonAPI/blob/master/quickstart/33-ego-drive-stepped.py) :
> [//github.com/lgsvl/PythonAPI/blob/master/quickstart/33-ego-drive-stepped.py](https://github.com/lgsvl/PythonAPI/blob/master/quickstart/33-ego-drive-stepped.py)
> 
> 如果使用Apollo，则需要使用Apollo 6.0或master并包括时钟传感器，以便Apollo可以使用上面脚本中所述的模拟时钟运行。
> https://www.svlsimulator.com/docs/simulation-content/sensors-list/#clock

Thank you

closed 2021-04-02 by cxf99