lgsvl/simulator #1324: Define sensor configurations in JSON file?

TrentWeiss, 2021-04-01:
In older versions of LGSVL, there was a JSON schema for creating sensor definitions and/or where those sensors are located with respect to the base_link (origin at centroid of the rear axle, X facing forward, Y facing left, Z facing up, just to drive that point home: #1308).

Is this possible in the new version?  The browser interface doesn't appear to have any way of creating/maintaining sensor suites other than the web editor, which is problematic.  Developing/creating new sensor configs on a large development team requires distributed version control tools (git, mercurial, etc.).  If the web-based GUI is the only way to build/maintain these sensor configs, it will be borderline impossible to maintain any kind of continuous integration and version control for the sensor definitions.

lemketron, 2021-04-01:
You can still access the raw json sensor configuration (though the format is new for 2021). 

There are screen shots and a description of how to find it here: https://www.svlsimulator.com/docs/user-interface/web/library/#migrating-legacy-sensor-configurations

Essentially:
[image]

and:
[image]

lemketron, 2021-04-01:
> Developing/creating new sensor configs on a large development team requires distributed version control tools (git, mercurial, etc.). If the web-based GUI is the only way to build/maintain these [...]

This is an excellent point. Thanks for the valuable feedback!