lgsvl/simulator #1326: Visual Scenario Editor with Apollo

summerpaul, 2021-04-02:
I created a VSE Json file , but the simulator did not run succesfully. I

[image]
[image]

EricBoiseLGSVL, 2021-04-02:
Sorry, we will need more information to debug this.  Did you use the Runner on Linux?  Did you have the assets cached?  What errors did you get in player.log?

summerpaul, 2021-04-06:
On ubuntu18 ,Error on latest run:Failed to launch TestCase runtime

summerpaul, 2021-04-06:
> Sorry, we will need more information to debug this. Did you use the Runner on Linux? Did you have the assets cached? What errors did you get in player.log? That image is too small to read the WebUI error

On ubuntu18 ,Error on latest run:Failed to launch TestCase runtime

EricBoiseLGSVL, 2021-04-06:
Thanks for the WebUI error, that helps.  Do you have the player.log from the unity application?  This shows more detailed errors in text format.
Did you also check the Known issue [here](https://www.svlsimulator.com/docs/running-simulations/running-simulator/#vsesimulation) at section 13?

ChengysTHU, 2021-11-11:
I have same issue, and my player.log file seems like this:

(Filename: /home/nics/2021.1/simulator/Assets/Scripts/Managers/TestCaseProcessManager.cs Line: 209)

Fallback handler could not load library /home/nics/Software/SVL2/simulator_Data/Mono/libKernel32
Fallback handler could not load library /home/nics/Software/SVL2/simulator_Data/Mono/libKernel32.so
Fallback handler could not load library /home/nics/Software/SVL2/simulator_Data/Mono/Kernel32
Fallback handler could not load library /home/nics/Software/SVL2/simulator_Data/Mono/libKernel32
Fallback handler could not load library /home/nics/Software/SVL2/simulator_Data/Mono/libKernel32.so
Fallback handler could not load library /home/nics/Software/SVL2/simulator_Data/Mono/libKernel32
Unable to launch app '/home/nics/Software/SVL2/simulator_Data/../TestCaseRunner/visualEditor/run': ApplicationName='/home/nics/Software/SVL2/simulator_Data/../TestCaseRunner/visualEditor/run', CommandLine='', CurrentDirectory='/tmp/LGElectronics/SVLSimulator/TestCaseRuntimeVolumes/438f4f4d-3141-44d5-b8c5-0e87198c23ee', Native error= Cannot find the specified file
UnityEngine.DebugLogHandler:Internal_Log(LogType, LogOption, String, Object)
UnityEngine.DebugLogHandler:LogFormat(LogType, Object, String, Object[])
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:LogError(Object)
Simulator.TestCaseProcessManager:StartProcess(String, IDictionary`2, String) (at /home/nics/2021.1/simulator/Assets/Scripts/Managers/TestCaseProcessManager.cs:227)
Simulator.Loader:RunTestCase(TemplateData) (at /home/nics/2021.1/simulator/Assets/Scripts/Web/Loader.cs:1029)
Simulator.<<StartAsync>b__0>d:MoveNext() (at /home/nics/2021.1/simulator/Assets/Scripts/Web/Loader.cs:469)
System.Runtime.CompilerServices.AsyncVoidMethodBuilder:Start(<<StartAsync>b__0>d&)
Simulator.<>c__DisplayClass41_0:<StartAsync>b__0()
Simulator.Loader:Update() (at /home/nics/2021.1/simulator/Assets/Scripts/Web/Loader.cs:297)
 
(Filename: /home/nics/2021.1/simulator/Assets/Scripts/Managers/TestCaseProcessManager.cs Line: 227)

Failed to start '1111111111111111111111111' simulation

Meanwhile, compared with released version, I don't have 'TestCaseRunner' folder after compilation.

EricBoiseLGSVL, 2021-11-11:
2021.1 has the runner locally in the binary we provide.
2021.3 release requires Docker so runner will work on windows and linux without needing to package with the binary.