lgsvl/simulator #1329: Logitech  G920 is not working with the simulator

rokon1014, 2021-04-02:
Hello,
I am having trouble using the Logitech G290 wheel controller to control a vehicle on the simulator.  I have done the following:

1. Added the wheel control sensor to the vehicle and used that sensor configuration to run the scene with Python API
[image]

2.  Checked the Input debugger in Unity editor. The unity editor is able to identify the device and it can read value. 
[image]

But I can not control the vehicle with the wheel controller. The simulator does not read any value from the device. It would be great if someone can help me to solve this. 

**Here is the player log file.** 
[Player-prev.log](https://github.com/lgsvl/simulator/files/6249455/Player-prev.log)

**I am using pythonAPI. Following is the code:** 

`
#!/usr/bin/env python3

from environs import Env
import lgsvl

env = Env()
sim = lgsvl.Simulator(env.str("LGSVL__SIMULATOR_HOST", lgsvl.wise.SimulatorSettings.simulator_host), env.int("LGSVL__SIMULATOR_PORT", lgsvl.wise.SimulatorSettings.simulator_port))
scene_name = env.str("LGSVL__MAP", lgsvl.wise.DefaultAssets.map_straight2lanesame)

if sim.current_scene == scene_name:
    sim.reset()
else:
    sim.load(scene_name)

spawns = sim.get_spawn()
state = lgsvl.AgentState()
state.transform = spawns[0]
print(spawns)
ego = sim.add_agent("764b0e15-7666-4b2c-beaa-d6a125f4b92f",lgsvl.AgentType.EGO, state)

sx = spawns[0].position.x
sy = spawns[0].position.y
sz = spawns[0].position.z
point = lgsvl.Vector(sx , sy, sz+50 )

state = lgsvl.AgentState()
state.transform.position = point
ego1 = sim.add_agent("764b0e15-7666-4b2c-beaa-d6a125f4b92f",lgsvl.AgentType.EGO, state)

ego.connect_bridge("127.0.0.1", 9090)
ego1.connect_bridge("127.0.0.1", 9091)

sim.run(time_limit=200)
`

EricBoiseLGSVL, 2021-04-02:
You should see input changes in the debugger.  This means your drivers / GHub is not installed correctly.  Once you have that fixed you will need to change the input names. We have found that Logitech changed the input names for the G920 wheel.  You will need to open SimulatorControls asset and reassign the inputs (or just replace the files below), then rebuild simulator.
Once Unity can recognize the device in the debugger, then you are all set to update inputs.

Here are the file changes needed:
[SimControls-20210331T214909Z-001.zip](https://github.com/lgsvl/simulator/files/6250635/SimControls-20210331T214909Z-001.zip)
This is for windows only, we haven't looked into Linux yet.

rokon1014, 2021-04-03:
> You should see input changes in the debugger. This means your drivers / GHub is not installed correctly. Once you have that fixed you will need to change the input names. We have found that Logitech [...]

Hi, 
Thank you very much. That worked like a charm. 

However, a small issue I faced. The acceleration action was inverted. When the acceleration pedal was not pressed it was providing the highest value and for full press the lowest. I edited the SimulatorControl asset in the unity editor and inverted the actions on keypress using the processor option.  Now it is working as it should.

closed 2021-04-03 by rokon1014

lemketron, 2021-04-03:
Yes, I found and fixed the accelerator inversion issue recently for the Logitech G920 on Windows. I plan to test the G29 and G920 on both Windows and Linux and share a fix for all.

Hopefully we can also move all of the necessary bits to the wheel controller plug-in to make it easier to support other controllers as well.

If anyone would like to look into implementing force feedback support, I'd love to integrate that as well!

lemketron, 2021-05-17:
> However, a small issue I faced. The acceleration action was inverted. When the acceleration pedal was not pressed it was providing the highest value and for full press the lowest. I edited the [...]

The fix for this in the WheelControlSensor would be to change the Windows line to match the Linux line here:
https://github.com/lgsvl/WheelControlSensor/blob/master/WheelControlSensor.cs#L85

That is, change the Windows line to this:
``accel = Mathf.InverseLerp(-1f, 1f, obj.ReadValue<float>());``

I'm looking to find out whether the latest WheelControlSensor plug-in includes this fix but if it doesn't you could build your own plug-in to fix this (or perhaps I can share mine)...