lgsvl/simulator #1330: How to control ego in real time with Python API？

cxf99, 2021-04-02:
How to control ego in real time with Python API？

EricBoiseLGSVL, 2021-04-02:
What do you mean?  Are you referring to throttle and steering commands with the python api or using an AD stack to control the vehicle?  In the python api quick start scripts, 05-ego-drive-in-circle.py, is an example of how to control the ego vehicle.

cxf99, 2021-04-03:
use api

cxf99, 2021-04-03:
If I use sim.run (3)  Then there will be catons in the simulator

cxf99, 2021-04-03:
sospensione

cxf99, 2021-04-03:
Pausa

cxf99, 2021-04-03:
sorry ， my  English is not accurate

EricBoiseLGSVL, 2021-04-05:
@cxf99 No worries, did the quick script help to see how to control the ego with API?  Yes, run(3) will run the simulation for 3 seconds and pause.

marcusvinicius178, 2021-04-07:
Hi @EricBoiseLGSVL  I would like to use Python API script with AD stack from AutowareAI, is it possible? Do you have a demo or some source code working? 

In addition I also would like to use Virtual Scene Editor (VSE) along with AutowareAI. I have defined my scenario with VSE, set AutowareAI bridge to work along it and published the simulation on SVL simulator. However the simulation opens and after some seconds close. I also would appreciate if the simulator kept opened and afterwards worked with AutowareAI AD stack. I mean use a ROS planner and set a nav_goal on rviz for the car move to this spot using Autoware_AI motion planner module. Could you assist me on that?

Thanks very much in advance

EricBoiseLGSVL, 2021-04-07:
@marcusvinicius178 are you using latest release? Please refrain from posting in multiple issues.  Create an issue and we will help as best we can

lemketron, 2021-04-14:
Please note that sim.run() is a blocking call and does not return until a timeout or callback occurs. And between calls to sim.run() the simulator is not running.

Please refer to #1323 for more information on passing a timeout to sim.run() (with an example) to run a stepped simulation.

A stepped simulation will run close to real time but it depends on how much python code you execute while the simulator is paused between each step.

closed 2021-07-06 by lemketron