lgsvl/simulator #1332: Missing some scripts/prefabs when I import release-2020.06 and CubeTown

rogeherlor, 2021-04-03:
I have cloned the repositories with Git LFS installed and made sure it is release-2020.06 and not the master branch. Unity version is 2019.3.15f1.
It is working and I can build the environment and LGSVL Simulator without errors, but in Unity I have the following errors:
[image]
Traffic Light prefab is missing
[image]
DestinationInfo is missing for pedestrians, so they don't move and an error appears
[image]
[image]

EricBoiseLGSVL, 2021-04-07:
Make sure you have the 06 release version of cubetown.  Looks like you have master or another branch

rogeherlor, 2021-04-09:
Thank you for answering.
I can only see the master branch on the repository https://github.com/lgsvl/CubeTown 
How can I download version 06 of cubetown?

EricBoiseLGSVL, 2021-04-09:
use the 2020.06 tag.  I recommend you move to our new release with many new features :)

rogeherlor, 2021-04-10:
Sorry but I don´t know why it doesn´t work for me. I think I am using the 2020.06 tag
[image]

EricBoiseLGSVL, 2021-04-12:
I recommend deleting the cubetown repo and re cloning.  Make sure unity is closed and then open after clone.  Then close Unity and open again.  If that doesn't work, try deleting the Unity Library folder and re open unity again.