lgsvl/simulator #1333: How to make the vehicle move backward through the keyboard control？

ideasplus, 2021-04-04:
I encountered a strange problem where I can't make the vehicle move backward through the keyboard control while other commands work well such as moving forward, truning left. When I press the ↓ button the car just slow down and then stop still.

lemketron, 2021-04-04:
You'll need to use the page-up and page-down commands to shift to forward and reverse, respectively.
https://www.svlsimulator.com/docs/user-interface/keyboard-shortcuts/

ideasplus, 2021-04-06:
> You'll need to use the page-up and page-down commands to shift to forward and reverse, respectively.
> https://www.svlsimulator.com/docs/user-interface/keyboard-shortcuts/

Thank you so much for your feedback。

closed 2021-04-07 by lemketron