lgsvl/simulator #1334: How to apply the scenario generated by Scenic in SVL Simulator and run Apollo 5.0 with it ？

201520906050, 2021-04-05:
Recently I used Scenic (a kind of hypothetical Python-like language) to generate some dynamic scenarios and ran them with SVL Simulator, but I found that there's no way to use Scenic Command to run the Scenic scripts in SVL Simulator with Apollo, and the scenarios can't be saved to local file system in order to reuse it next time. The scenarios are generated only when the terminal is running，when it‘s closed nothing will be left.

lemketron, 2021-04-05:
To use Scenic you would configure the SVL Simulator to use "API Only" simulation as documented here:
https://www.svlsimulator.com/docs/running-simulations/running-simulator/#apionlysimulation

You can read more about using Scenic with SVL Simulator here:
https://scenic-lang.readthedocs.io/en/latest/simulators.html#lgsvl

marcusvinicius178, 2021-04-07:
Hi does the SCENIC works for AutowareAI bridge too and for SVL simulator or just for LGSVL?

lemketron, 2021-04-07:
Hi @marcusvinicius178,
Scenic currently works only with Apollo. It uses the "dreamview" library that is part of the [PythonAPI](https://github.com/lgsvl/PythonAPI) (in master for 2021.1 simulator, and in "[preview2](https://github.com/lgsvl/PythonAPI/tree/preview2)" tag for 2020.06 simulator).

201520906050, 2021-04-09:
> To use Scenic you would configure the SVL Simulator to use "API Only" simulation as documented here:
> https://www.svlsimulator.com/docs/running-simulations/running-simulator/#apionlysimulation
> 
> You can read more about using Scenic with SVL Simulator here:
> https://scenic-lang.readthedocs.io/en/latest/simulators.html#lgsvl
> 
> Scenic will communicate with SVL Simulator through the LGSVL Python API as well as with Apollo using the Dreamview library which is part of the Python API.

But Scenic only works with the Scenic command, and the Scenic syntax isn't compatible with python api. Scenic is just a kind of hypothetical Python-like language, its syntax is similar to python, but it doesn't support python interpreter, so you can't write your Scenic code and Python code together. And the Scenic script names are end with ".scenic" instead of ".py". So the way using "dreamview" library in Scenic language to control Apollo doesn't work.

lemketron, 2021-04-14:
Linking with related issue in Scenic project: https://github.com/BerkeleyLearnVerify/Scenic/issues/31

lemketron, 2021-04-14:
> But Scenic only works with the Scenic command, and the Scenic syntax isn't compatible with python api. Scenic is just a kind of hypothetical Python-like language, its syntax is similar to python, [...]

The first version of Scenic generated static scenarios (waypoints for vehicles to follow) but the current release supports dynamic scenarios where vehicle state information can be checked during the simulation to influence behavior during the scenario. In this mode the Scenic runtime communicates with SVL Simulator using the Python API and would step the simulation while this is taking place.

The SVL Simulator will ideally get a new runtime template for Scenic scripts and will invoke the Scenic runtime with the (dynamic) abstract scenario to be executed. That integration is not finished yet, so in the mean time we suggest running the simulator in "API-Only" mode, and running Scenic from the command line. It will communicate through the Python API, run the desired scenarios, and exit.

Scenic already has integration with LGSVL Simulator 2020.06 (though it likely needs to be updated for SVL Simulator 2021) and supports the Dreamview library that is now part of LGSVL Python API as well. You can see that code here:
https://github.com/BerkeleyLearnVerify/Scenic/blob/master/src/scenic/simulators/lgsvl/simulator.py#L79