lgsvl/simulator #1335: Real world positioning coordinate input

yufeilongyuan, 2021-04-05:
Hi team：

             I'm very excited to find this project. Thank you for your hard work. I have a question to ask. I have real coordinate data (in TXT format). I want to import LGSVL, then edit and modify the positioning data. Finally, output the data in BaiDu Apollo format. Can you give us some guidance or some references? Hope to get your reply. thank you!

EricBoiseLGSVL, 2021-04-07:
Thanks and we are grateful you are liking what we have built.  What is the "real coordinate data"?  Is this a HDMap?  Point cloud data?  SVL Sim lets you import HDMap data, and the coordinates are applied to the MapOrigin.cs object.  This data can also be used to create a mesh but it is still a work in progress.  If you have point cloud data you can import this instead of making annotation meshes.

yufeilongyuan, 2021-04-08:
@EricBoiseLGSVL I'm glad to hear from you. Maybe my description is not very accurate, which brings you trouble in understanding. Now, I'll describe the details in detail and hope to get your help. I can get GPS data through **APOLLO** vehicles, and generate base_map.txt Or base_ map.bin File through apollo's own scrip, which can be used in Apollo system later. But for some reason, I want to edit the base_ map.bin in **LGSVL**, but I can't import it. There are always some problems (as shown in the figure). Some of my thoughts: does this have anything to do with the Apollo version? I use the official version 5.0 code. Do I need to use your Apollo 5.0 code? What's the difference?I have encountered great difficulties, and I am not familiar with C# and Unity, and I don't know how to solve it. I hope to get your help. Thank you very much

EricBoiseLGSVL, 2021-04-08:
@yufeilongyuan Are you able to post the base_ map.bin here?  If not you will need to place debug logs at ApolloMapImporter.cs line 78 to see what is causing the error from the data you have.  The issue shouldn't be from apollo version.  Is the hdmap large? complex intersection?  Our importer might not work with an edge case that is happening in your data.  This is the latest simulator release right?

yufeilongyuan, 2021-04-09:
@EricBoiseLGSVL I've solved this problem, because base_map.txt,there is no "head" structure in the file, which leads to the program reading data error. The root cause is the lack of relevant processing in Apollo script. By the way, I can import hdmap, but I still don't know how to edit it. do you have any related information? Official documents are not very helpful to me, thank you.

EricBoiseLGSVL, 2021-04-09:
Excellent, glad you were able to solve.  If you have any more details on your fix please post here for others that have the same issue as you.

Sorry the annotation documentation is not helpful.  Can you be more clear on the reason why?  Annotation that supports simulator NPC navigation and exporting to 4 different HDmap formats makes for a complex feature.  What would make it more helpful?

yufeilongyuan, 2021-04-09:
no problem! I am very familiar with Apollo system and hope to help you. About map annotation, I may need to explore. There is not much systematic information on the Internet.

EricBoiseLGSVL, 2021-04-09:
Sounds good.  Having someone familiar with Apollo is always helpful.  Feel free to answer any issues you see and you know the solution.  As for annotation, just be aware that annotation in svl sim is a hybrid of HDMap data that we use to export to HDMaps.  It also is used by NPCs and pedestrians to navigate during a simulation.  The workflow is import HDMap, make adjustments to accommodate NPC behavior. It is not a tool to strictly edit HDMaps, edit only if you want alter the map from real world or the map is fake.  Otherwise you would use the original map for the AD stack.

yufeilongyuan, 2021-04-09:
OK, if I come across some questions about apollo, I will try my best to answer them. As for map annotation, after a period of time when I am not busy with my work, I want to try to develop a map editing tool. Lgsvl is a very excellent tool, and I hope it will get better and better. Thanks again, I'll close the issue.

closed 2021-04-09 by yufeilongyuan