lgsvl/simulator #1336: Problem adding NPC vehicle

rokon1014, 2021-04-05:
Hello, 
I am trying to add an NPC vehicle on a simulation started with PythonApi. To be sure, I am running the script "08-create-npc.py" from the quickstart folder and getting an exception thrown. Following is the error message: 

```
Python API Quickstart #8: Creating various NPCs
Traceback (most recent call last):
  File "single_car.py", line 38, in <module>
  sim.add_agent(name, lgsvl.AgentType.NPC, state)      
  File "C:\Users\Rokon\AppData\Roaming\Python\Python37\site-packages\lgsvl\utils.py", line 23, in new_f
    return f(*args, **kwargs)
  File "C:\Users\Rokon\AppData\Roaming\Python\Python37\site-packages\lgsvl\simulator.py", line 151, in add_agent
    uid = self.remote.command("simulator/add_agent", args)
  File "C:\Users\Rokon\AppData\Roaming\Python\Python37\site-packages\lgsvl\remote.py", line 69, in command
    raise Exception(data["error"])
Exception: Object reference not set to an instance of an object

```
What am I missing here? I am on windows 10 and built the simulator from the source. Any suggestion would be appreciated.

EricBoiseLGSVL, 2021-04-07:
Did you add the NPC to the simulator?  Is the NPC in the NPCSettings.asset, NPCManager.prefab?

Jiaqing-Cao, 2021-06-01:
@EricBoiseLGSVL , I met same problem too, 
Even using the 2021.1.1 release exe file, the quickstart code shows the above error.
Could you be more specific about how to add the NPC to the simulator?

EricBoiseLGSVL, 2021-06-02:
Can you be more specific in what the issue is?  Are you just trying to run the api quick script with existing npc's or trying to add a new npc?

Jiaqing-Cao, 2021-06-02:
I am just trying to run the "10-npc-follow-the-lane.py: quick example with the release exe file one windows 10, 
it will raise error at line:

npc1 = sim.add_agent("Sedan", lgsvl.AgentType.NPC, state)

Exception: Object reference not set to an instance of an object

EricBoiseLGSVL, 2021-06-03:
I have tested on windows both 08 and 10 quick scripts without errors.  Do you have the correct EGO vehicle that the quick scripts referenced in your library as well as Borregas Ave environment?  Also what is the single_car.py script referenced in the callstack?  Also, are you using API Only simulation or pasted in WebUI?  I recommend using APIonly and running script in powershell.  See [here](https://www.svlsimulator.com/docs/running-simulations/running-simulator/#apionlysimulation).

Jiaqing-Cao, 2021-06-04:
I was using the "python API" mode and running script in powershell before, after changing into "API Only', it works correctly. Thank you. @EricBoiseLGSVL

EricBoiseLGSVL, 2021-06-04:
Excellent, seems we have a bug with WebUI PythonAPI mode.  We hope to have a fix soon.  Thanks for posting the issue