lgsvl/simulator #1337: Best way to create custom race tracks for the simulator?

sisaha9, 2021-04-06:
I see some packages like EasyRoader3d or RoadRunner don't have direct support in terms of meshes for SVL. Is there a recommended way to go about creating tracks in the free version of SVL?

EricBoiseLGSVL, 2021-04-07:
SVL's issues with RoadRunner are: (not sure on EasyRoads but it's prob similar issues)
1) Roadrunner doesn't support HDRP.  You will need to convert materials to the Unity standard shader and then import into svl sim and convert to hdrp. see #1074 and #1296
2) Road meshes from Roadrunner for are multiple meshes with transparency so our shader for wetness/damage/segmentation won't work.  We require one mesh with one material.  If you don't need this feature then you can import as is.
3) We don't support the traffic lights from Roadrunner, if you need traffic lights, you must use our system.

We also have the feature to create meshes from annotation hdmap data but it is very simplistic.

Hope this helps.

sisaha9, 2021-04-07:
@EricBoiseLGSVL Am I able to make a track in Blender and export to SVL? Any thing I need to be wary off? I was planning to do something like this tutorial: https://youtu.be/1I9UtB1YJAs

EricBoiseLGSVL, 2021-04-08:
@sisaha9 Sure, Blender is a great tool.
1) Blender axis are different than Unity.  Unity has Z forward and Y up X+ right.  Just search export to unity from blender if you have issues
2) Export as Fbx
3) Embed the textures in fbx.  Unity has options to extract when you import
4) Make the road one mesh with one material texture to use the svl sim environmental effects shader.
5) If you want segmentation be sure to separate meshes for sidewalk, road, etc.  That allows different materials to be applied to each type.

That looks like a good tutorial.  Let us know if you have issues, I can ping our artists.  Good luck and can't wait to see the track!

sisaha9, 2021-04-08:
Thanks! I will close this issue now since it was about good ways to create assets/worlds for SVL. I will open up a new issue specific to Blender if I encounter any issues. Thanks!

closed 2021-04-08 by sisaha9