lgsvl/simulator #1338: How can I run simulator behund a proxy?

DrShIkIgAmy, 2021-04-06:
I wanna run simulator on Ubuntu, so Proxifier is not a solution. LGSVL doesn't take system settings (proxy at least), that's why it stucks when I run simulator. What should I do?

lemketron, 2021-04-06:
What version of simulator are you running? 2020.06 or 2021.1?  
Have you tried proxifier or privoxy as mentioned in #360?

DrShIkIgAmy, 2021-04-06:
> What version of simulator are you running? 2020.06 or 2021.1?
> Have you tried proxifier or privoxy as mentioned in #360?

I'he tried both of them. I've seted up privoxy to forward all requests, but it worked on browser only. Simulator still knock on server bypassing privoxy.
P.S. M.b. I'm a bit stupid and I did it wrong