lgsvl/simulator #1339: Wrong output of LiDAR-based perception module triggers undesired behavior

ideasplus, 2021-04-06:
I have encountered a problem that has been bothering me for a long time. When I run the lidar-based perception module and other downstream modules with simulator,  perception module sometimes output the **_non-existent_** obstacle that is located very close in front of the ego car, which will trigger the braking decison as shown in the figure.

[image]

Please look through the scenarios with recording video for more details. 
https://drive.google.com/file/d/1Ludax7yI0vAHVgIo04G1q2nsrmPpiwMj/view?usp=sharing

Note that I'm using Apollo 6.0 master and LGSVL Simulator 2020.06. I can also see the same phenomenon in other old versions of apollo such as Apollo 5.5.

I’m not sure if it’s the problem of the model itself or the simulator and look forward to the solution.

lemketron, 2021-04-07:
You may wish to ask about this on the Apollo project...
https://github.com/ApolloAuto/apollo/issues

lemketron, 2021-04-14:
Linking to related issue: https://github.com/ApolloAuto/apollo/issues/13669