lgsvl/simulator #1340: Visual Scenario Editor with Apollo5.0 not working on Ubuntu 18.04

Jiaqing-Cao, 2021-04-07:
Hi LGSVL team,

I download the latest 2021.1 release Linux executable file and try to use Visual Scenario Editor with Apollo5.0 according to the demo video on your website.

However, the simulator just show the environment and vehicle for like one second and stop.
[image]

The test report did not give useful info.

And I attach the Player.log:
[Player.log](https://github.com/lgsvl/simulator/files/6268213/Player.log)

I look through the log and see
[ERROR][dreamview] HD Map BorregasAve was not set.

But I download the latest apollo5.0 HD map from https://wise.svlsimulator.com/ library map.
And use generate_map.sh to get the following files correctly:
[image]

Could you help me to see which part causes error?

marcusvinicius178, 2021-04-07:
Hi I had the same issue, but for AutowareAI bridge. 
I do not believe that your issue is regarding the environment you dowloaded.
The same thing happen to me when trying to use VSE_AutowareAI, the simulator opens and closes
[image]

It says always the following messages: 
"Error on latest run: Bridge socket was unexpectedly disconnected"
"Busy: Another Simulation is currently running on this cluster"

I know my cluster is not busy, because I have deleted all the old ones and recreated one new cluster from scratch...

With Python API option, the simulator also open and closes. **HOWEVER this can be solved with a trick on python** script, that is just let the simulation running infinitely pasting the following line at the end of python script code:
sim.run(0)

But for VSE as there is not this option, I also would like to know how to let the simulator opened without close. If you know please trigger me also. Thanks

@Jiaqing-Cao how do you get the Player log to attach?

Jiaqing-Cao, 2021-04-07:
@marcusvinicius178  the Player.log is at :
/home/xxx/.config/unity3d/LGElectronics/SVLSimulator/ on my machine.

DrShIkIgAmy, 2021-04-08:
> @marcusvinicius178 the Player.log is at :
> /home/xxx/.config/unity3d/LGElectronics/SVLSimulator/ on my machine.

I have the same trouble. But It works for me about 2-3 seconds, passes a few distance and closes

Jiaqing-Cao, 2021-04-09:
@EricBoiseLGSVL 

Could you help me to look into this issue?

EricBoiseLGSVL, 2021-04-09:
The error is here in player log.
2021-04-07 00:44:53,665 [ERROR][dreamview] HD Map BorregasAve was not set. Verify the files exist in /apollo/modules/map/data/BorregasAve and restart Dreamview -- Aborting..

Be sure Apollo is getting BorregasAve not borregasave.  We had a recent name change and it might not have been updated.  We will look into this issue.  @KarByc didn't you have this same issue?  Can you post here with your solution please, thanks.

EricBoiseLGSVL, 2021-04-09:
another thing to check is that you have BorregasAve in your library and the name is the same.

Jiaqing-Cao, 2021-04-09:
@EricBoiseLGSVL ,

Yes, you are correct, I changed the HD map folder name in Apollo into "BorregasAve", it works, thanks about that.

However, I met a new problem.
The Scenario will stop after about 23 seconds, but the ego vehicle did not reach the destination point, and NPC did not reach their waypoints either. the playback in VSE shows it will run about 57 seconds.
The test report shows success though.

Do you have any ideas about that?

EricBoiseLGSVL, 2021-04-09:
yes this is a bug and we hope to have a fix soon. Marking issue as a bug and will post with updates

KarByc, 2021-04-09:
@EricBoiseLGSVL it should wait at least for the NPCs to reach their destination waypoints.

@Jiaqing-Cao Can I ask for the Player.log?

Jiaqing-Cao, 2021-04-12:
@KarByc 

I attached the Player.log
[Player.log](https://github.com/lgsvl/simulator/files/6293093/Player.log)

KarByc, 2021-04-12:
@Jiaqing-Cao thanks for the Player.log.
One race condition was found and we are working on fixing it.