lgsvl/simulator #1342: Busy Cluster: VSE + AutowareAI Bridge Simulation fails on SVL

marcusvinicius178, 2021-04-07:
Hi LG staff:

Could you assist me on cluster issues here please?

I set up a custom scene on VSE according to tutorial and video documentation (https://www.svlsimulator.com/docs/creating-scenarios/visual-scenario-editor/)

The first time worked (both for Apollo and AutowareAI bridge). And I tried to integrate AutowareAI modules with it, but failed as show the image below:

[image]

However after some hours I tried again the same simulation and I had one warning before launch the simulation:
**"Busy: Another Simulation is currently running on this cluster"**
 Even with this warning the simulation opened, but quickly closed after some seconds:

[image]
Then after try again to launch the simulation I have seen the error on SVL icon interface:
 **"Error on latest run: Bridge socket was unexpectedly disconnected"**
[image]

I did not find a player.log (where is it located). Then I will post the test result (error):
[image]

In order to fix, I  removed all my previous created clusters and all simulations as well on SVL website. 
Started from scratch the scenario setup and Simulation setup. However the error persists... 
My step-by-step to set the VSE and Simulation on SVL are below (with images)

A)VSE:
[image]

B) Simulations Edit

[image]
[image]

So at first I would like your assistance to:
1) Fix VSE - AutowareAI Simulation cluster communication issue (as output maintain the simulation opened and running)
2) Fix VSE - AutowareAI modules integration to work with the custom scene created by VSE (as it works without this integration on this demo https://www.svlsimulator.com/docs/system-under-test/autoware-instructions/)

Thanks very much on advance!!!

hadiTab, 2021-04-07:
@marcusvinicius178 see the documentation on [reading logs](https://www.svlsimulator.com/docs/running-simulations/python-runtime/#reading-logs) for `Player.log`.

As for the message "Busy: Another Simulation is currently running on this cluster", it tends to appear when the simulation is loading. It can be safely ignored.

The error about the bridge socket ("Error on latest run: Bridge socket was unexpectedly disconnected") likely indicates that the bridge was not connected. When starting a simulation, it is expected that the bridge is already running. Otherwise there will be an exception.

As for the errors you are seeing in the AutowareAi terminal; those indicate that the `mission_planner` was not able to find a valid path between the current vehicle location and the specified destination. There can be several reasons for this:

1. It is not possible to get to the specified destination from the present starting point. Note that lane changing is disabled in AutowareAi by default (and last I checked did not work properly when enabled) so make sure your destination is accessible without a lane change.

2. There is an issue with localization, so AutowareAi thinks the ego vehicle is somewhere else and can't find a valid route to the destination. Check in rviz and make sure the vehicle appears to be localized correctly.

3. After giving an invalid destination, AutowareAi will reject all subsequent destinations even if they are valid (it's a bug). To get around this disable and re-enable the mission planner throught the runtime manager by pressing the mission planner button and then try a new destination.

marcusvinicius178, 2021-04-07:
Hi @hadiTab thanks for the answer.
As far as I understood AutowareAI work partially with VSE and simulator.
Do you know if the same issues you wrote above happen with the PythonApI-AutowareAi integration? (Lane change, localization, etc?) Because from the demo codes (without PythonApi or VSE integration) all the packages work properly. Then I still did not understand if this issue you said is due to integration or just an AutowareAI issue???

About the socket error connection, I still did not understand why it happened. Because as I said the first time I set the VSE with AutowareAI or with Apollo and played the simulation, the simulator opened. However after other attempts hours later the same simulation did not work.
When you say that the bridge is expected to be running before, I imagine that I must run a script to connect the AutowareAI and SVL simulator, as it is required with Ros2-web-bridge with AutowareAuto? In AutowareAuto I always launched Ros2-web-bridge or lsgsv-bridge before start the simulator....However for AutowareAI I did not see on documentation any necessary procedure regarding bridge or still a bridge package to download and launch. Is there one pkg to launch linking AutowareAi and SVL simulator??

hadiTab, 2021-04-09:
@marcusvinicius178 The lane change functionality is an issue with AutowareAi as far as I know and not related to the simulator. Using PythonApi instead of VSE would not change anything in that regard.

As for partial support for AutowareAi and VSE; you should be able to set up a simulation and run it but you will need to manually startup AutowareAi, make sure all of it's modules are working properly and manually set a destination for the ego vehicle. This can be an issue with VSE simulations because the simulation will not wait for AutowareAi to be ready before playing through. 

As for the socket error; when you are setting up AutowareAi one of the launch files we provide is called `my_sensing_simulator.launch` which includes a command to start `rosbridge`. This launch file must be run before starting the simulation. Another option is to create an interactive simulation; which places a play button at the bottom of the screen. In that case you can start the simulation but not hit play until you have the bridge running. That would also work.

marcusvinicius178, 2021-04-22:
@hadiTab  Thanks for the info. To close this issue I have more 2 doubts in advance:

1 - Is possible to add the Lexus with PythonAPI for AutowareAI ? I tried to add in python script with the name:

`ego = sim.add_agent(env.str("LGSVL__VEHICLE_0", lgsvl.wise.DefaultAssets.ego_lexus2016rxh_autowareai), lgsvl.AgentType.EGO, state)`

But I got an error...This vehicle is not available in wise.py in https://github.com/lgsvl/PythonAPI/tree/master/lgsvl/wise
Then I tried to add a name to it within its Autoware_AI id:

`ego_lexus2016rxhybrid_autowareai= "fbeb6e0f-6020-4996-87a5-d56f907ba1ed"`

However this attempt did not work. **Is it possible to call this vehicle using his ID tag (= fbeb6e0f-6020-4996-87a5-d56f907ba1ed) in Python script?** If yes how?

I am trying this approach because the Jaguar vehicle is not working with the demo (BorregasAv) provided in the documentation. I mean it is possible to launch the Jaguar vehicle by the use of PythonAPI, but the demo tutorial https://www.svlsimulator.com/docs/system-under-test/autoware-instructions/ the AutowareAI does not work with this car (work just with Lexus). For that I would need to modify the launch files, specifically vehicle_description package, and I am still not sure if would work, considering the launch file launches a "general" vehicle model, not specific ones. Besides that I am having a lot of problems to commit Autoware_AI image in the running container: https://stackoverflow.com/questions/67185333/modify-the-docker-image-and-save-it-for-not-loose-the-content-when-building-the/67198856?noredirect=1#comment118790600_67198856

2) Can the python script (Python Api application) be integrated with the AD stack in such way that is synchronized with the launch files. I mean it would be also good **if possible to call this scripts that launch the LG agents to the simulator, from a ROS Launchfile, Is this possible or would not work?**

 I think in add a python script to add the agents to the scene inside a launch file, then the script would be triggered when launching a module and would be for example more synchronized, or not?

**Do you suggest some solution for Python API synchronization with AD stack?** If I add a static pedestrian there is no problem but with move it would be an issue right? I have also read documentation that you can "hold" the agents until the ego vehicle achieve some desired spot, and then the agent start its moves. Is this feature working with PythonApi or just VSE? **Are there tutorials how to use PythonAPI within Autoware_AI AD stack synchronized?**

Thanks in advance

marcusvinicius178, 2021-05-18:
It was not possible to add Lexus vehicle. However to use the Jaguar with Autoware.AI was possible and enough to generate simulations with PythonAPI agents:

https://www.youtube.com/watch?v=XSKYE1Oxq-c&list=PL_d_YiRmUA3nnTRmF8QJKyPoEQhrdvpP2&index=9&t=653s

Therefore I am going to close this issue.

closed 2021-05-18 by marcusvinicius178