lgsvl/simulator #1343: After the latest update ,  controling car through the old msg interface is not working

sunjay-lee, 2021-04-07:
As the title shows , after the latest update,I found it no-use to control the car by the old msg interface. When I gave it a brake value(in the old 2020.06 version ,I achieved this by give the numeric field called "vehicle_cmd.ctrl_cmd.linear_acceleration" a negative value) it really worked. But now in the 2021.01 version, when a negative value is given the car is still accelerating.
As shown in the picture , I can know how it works in the old version , while I can not find this code period or relevant period in the new version. I wonder that is there any solution or a new command code period and how it works now in the latest version. [picture_url](https://static.dingtalk.com/media/lALPDiCpulZWDXDNAgnNAqs_683_521.png_620x10000q90g.jpg?auth_bizType=IM&auth_bizEntity=%7B%22cid%22%3A%22175093251%3A2126240356%22%2C%22msgId%22%3A%226771360699962%22%7D&bizType=im&open_id=2126240356)

hadiTab, 2021-04-07:
@sunjay-lee can you provide more information on how you are setting up your simulation? IIRC, in 2020.06 we only supported controlling through `vehicle_cmd.twist_cmd` and not `vehicle_cmd.ctrl_cmd`. To use the latter, make sure that you are also publishing a gear value of 64.

sunjay-lee, 2021-04-07:
> @sunjay-lee can you provide more information on how you are setting up your simulation? IIRC, in 2020.06 we only supported controlling through `vehicle_cmd.twist_cmd` and not `vehicle_cmd.ctrl_cmd`. To use the latter, make sure that you are also publishing a gear value of 64.

[image]

Well, in the 2020.06 version , as the code in the picture shows , the car read the braking value from the vehicle_cmd.ctrl_cmd , so I use the period to give the car a brake value.

hadiTab, 2021-04-07:
It seems like there is a bug in the braking for autoware ai. We will look into this.