lgsvl/simulator #1344: Bridge client missing when running Apollo master with SVL 2021.1

kesav-v, 2021-04-07:
When I try to run the [Dreamview example](https://www.svlsimulator.com/docs/python-api/dreamview-api/#example) with the SVL 2021.1 simulator and the latest Apollo built from master, I get the following error:

`Exception: Agent 'b7a848fd-2722-4360-b6a8-51e70c4776f5' is missing bridge client`

I'm using an API Only simulation and this is the error that comes up when the script tries to run `ego.connect_bridge(BRIDGE_HOST, BRIDGE_PORT)`. The ROS bridge is running on the Apollo side, and I was able to manually create a simulation in BorregasAve and set a destination in the Dreamview UI. Any idea what might be causing this issue? Thanks in advance!

marcusvinicius178, 2021-04-07:
Hi I am not an expert in Apollo, just worked with ROSand Autoware, however I believe you should not have a ROS bridge to work with Apollo, instead you must have an Apollo 5.0 or Apollo 6.0 bridge set in Autopilot tab, right?

kesav-v, 2021-04-07:
Yes, I just meant that I had `bridge.sh` running from the Apollo Docker container. I didn't see any options available in the Autopilot tab using an "API Only" simulation - but when I ran a manual simulation in the UI using BorregasAve I did set Apollo 6.0 bridge in the Autopilot tab and I was able to get that working.

lemketron, 2021-04-07:
For "API Only" simulation, the specific vehicle (and sensor configuration) and associated bridge is connected by the Python script as you can see here:
https://github.com/lgsvl/PythonAPI/blob/master/quickstart/33-ego-drive-stepped.py#L35
https://github.com/lgsvl/PythonAPI/blob/master/quickstart/33-ego-drive-stepped.py#L40

kesav-v, 2021-04-07:
Ah I see, so the difference is the way I added the vehicle. I had to do `ego = sim.add_agent(lgsvl.wise.DefaultAssets.ego_lincoln2017mkz_apollo5, lgsvl.AgentType.EGO, state)` instead of `ego = sim.add_agent('Lincoln2017MKZ', lgsvl.AgentType.EGO, state)`. But now I'm running into an issue where I'm able to see the vehicle move, but it gets stuck at the intersection and doesn't go through even when there's a green light. Is that expected behavior? Looking in Dreamview, the thick blue line that shows the planned trajectory only goes up to the beginning of the intersection even though the route shown by the red line goes through the intersection.

hadiTab, 2021-04-07:
@kesav-v is the traffic light perception in Apollo working? Is it getting detected as green in dreamview?

lemketron, 2021-04-07:
> lgsvl.wise.DefaultAssets.ego_lincoln2017mkz_apollo5

Are you using [modular testing](https://www.svlsimulator.com/docs/tutorials/modular-testing/)? If so, then you should be using `ego_lincoln2017mkz_apollo5_modular` which enables the ground truth and traffic signal sensors.

kesav-v, 2021-04-08:
@lemketron Thank you! That fixed the issue.

closed 2021-04-08 by kesav-v