lgsvl/simulator #1345: Same angularVelocity sent from LGSVL shows up as 2 different values ImuSensor vs GpsOdometrySensor

henrychoi, 2021-04-07:
Hello, using 2020.06 branch still.
I am seeing a significant difference between the nav_msgs::msg::Odometry's twist.twist.angular and sensor_msgs::msg::Imu's angular_velocity.   On the Unity side, both virtual sensors use the RigidBody.angularVelocity in FixedUpdate method as the ground truth source.  The angular velocity from the IMU seems correct (in rad/s), and the rate in the twist message seems significantly lower than the IMU value (although it does seem to respond!).  I compared the ratio between the two of them, but there doesn't seem to be a fixed ratio, as you can see here
(gdb) p 0.187785 / 0.014633
$3 = 12.832980250119594
(gdb) p 0.408175 / 0.019911
$4 = 20.499974888252723

Any thoughts?  Thank you for reading.

hadiTab, 2021-04-07:
@henrychoi which bridge are you using in your sensor configuration?

henrychoi, 2021-04-07:
I am using the default bridge that comes with the Ubuntu 20.04-ros-foxy.  I upgrade automatically, and it should be the latest.

hadiTab, 2021-04-09:
From a quick glance at the code I can't see what's wrong. We'll need to take a closer look at it.