lgsvl/simulator #1348: svlsimulator-linux64-2021.1与Autoware1.14联合仿真网页无法接收topic / vehicle_cmd

Stuare8, 2021-04-08:
svlsimulator-linux64-2021.1
仿真器Info
中报错如下图所示：

[image]
[image]
[image]
[image]
[image]

EricBoiseLGSVL, 2021-04-08:
Hello, please describe your issue in english.  Also post your player.log so we can see the stack trace or grab the details from the first null error and trace and post here.

Stuare8, 2021-04-09:
@EricBoiseLGSVL The problem has been solved.Thank you

EricBoiseLGSVL, 2021-04-09:
Excellent, glad you were able to solve

closed 2021-04-09 by EricBoiseLGSVL