lgsvl/simulator #1349: About the usage of radar in the latest version

sunjay-lee, 2021-04-08:
In the latest version(2021.01), I find  using radar through the lgsvl_msgs and bridge does not work as it in the 2002.06 version . As the 1 part of the picture shows , after I follow the instructions at [your document](https://www.svlsimulator.com/docs/system-under-test/lgsvl-msgs/), there is a error in publishing the relevant radar topic . But in the 2020.06 version, the following the same steps did work.
As for the second problem, I find it is similar to the [issue1219](https://github.com/lgsvl/simulator/issues/1219#issue-788452448), there is an exception when using radar in the 2021.01 version tells"NullReferenceException: Object reference not set to an instance of an object" and there is not any change of your source code .
[image]

EricBoiseLGSVL, 2021-04-08:
Linking to #1219 and we will look into this, thanks for the issue.

sunjay-lee, 2021-04-09:
> Linking to #1219 and we will look into this, thanks for the issue.
Well, as the picture shows, I add a line of code to new a list, and that does work to solve the bug 2. I guess you may forget to initialize the list before. And there remains bug 1 that there is still errors in publishing topic and the exception does not even tell the topic channel.

[image]