lgsvl/simulator #1350: Pause simulation anytime during sim.run()

DeepCollision, 2021-04-08:
Hello team, is there any way to pause the simulation process when the simulator is running?
For example, when i call the api sim.run(), and i want to pause the simulation at a certain time, can i do this?

lemketron, 2021-04-08:
Please note that sim.run() is a blocking call and does not return until a timeout or callback occurs.

Please refer to #1323 for more information on passing a timeout to sim.run() (with an example) to run a stepped simulation.

closed 2021-04-08 by lemketron

DeepCollision, 2021-04-09:
> Please note that sim.run() is a blocking call and does not return until a timeout or callback occurs.
> 
> Please refer to #1323 for more information on passing a timeout to sim.run() (with an example) to run a stepped simulation.

But is that reasonable by doing that, for sometimes we do not know when to pause the simulation, in other word, we do not know how long we should let the sim.run() last. 
I just want to pause anytime i want.

lemketron, 2021-04-09:
If you want to pause the simulation you can use the pause button in the UI:
https://www.svlsimulator.com/docs/user-interface/simulation-menu/#play-pause-button

Otherwise, you'll need to either pass a timeout in sim.run(), or wait for a callback to be triggered, or use [stepped simulation](https://github.com/lgsvl/PythonAPI/blob/master/quickstart/33-ego-drive-stepped.py) to step the simulation at close to real-time while (programmatically) checking for whatever condition would make you wish to pause the simulation.

DeepCollision, 2021-04-09:
> If you want to pause the simulation you can use the pause button in the UI:
> https://www.svlsimulator.com/docs/user-interface/simulation-menu/#play-pause-button
>
> Otherwise, you'll need to either pass [...]

Thanks, "If you want to pause the simulation you can use the pause button in the UI:
", so there is no API can directly do this, and sim.run() can not be interrupted unless timeout, right?