lgsvl/simulator #1351: Is there any way to read the image by camera without saving it to the disk?

DeepCollision, 2021-04-08:


DeepCollision, 2021-04-08:
Hello team, I found that if i want to get an image and process it, firstly, i need to call camera.save() to save the image to disk then read the imgae from disk and process it, which is time consuming. 
Is there any other way to do that without saving first?

lemketron, 2021-04-08:
The best way to capture image data and process it in real time would be to use a sensor bridge (such as ROS, ROS2, or CyberRT) and subscribe to image data with a component that connects to that bridge.

DeepCollision, 2021-04-10:
> The best way to capture image data and process it in real time would be to use a sensor bridge (such as ROS, ROS2, or CyberRT) and subscribe to image data with a component that connects to that bridge.

Thanks for your reply, I have connect to Apollo CyberRT and get the image data from the bridge, but I found the image data encoding is rgb8 and I want to decode it into opencv format, is there any way can do that?
I asked a similiar question in Apollo respo, see [Apollo issue](https://github.com/ApolloAuto/apollo/issues/13676#issue-854600646)

closed 2021-07-16 by EricBoiseLGSVL