lgsvl/simulator #1353: Problem about controling Jaguar2015XE via keyboard in CubeTown in Unity3D.

wobushigaoying, 2021-04-09:
Hello, I followed the instruction of "building XXX from source", and that content did work correctly.

I then cloned CubeTown, Jaguar2015XE, and KeyboardControlSensor from Github. I put all these assets in the right"External" folder.

I dragged the CubeTown prefab into the Hierarchy tab in Unity3D, as well as the Jaguar2015XE prefab and SimulatorManager prefab. I also dragged the KeyboardControlSensor prefab into Jaguar2015XE as a sub-component in the Hierarchy tab. However, the simulation did not work after I click RUN. Here are the errors:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

All these errors seem to refer to only one type of error, that is "NullReferenceException". I checked all the relevant scripts and prefabs, and I found all the references were actually existed and cited correctly. However, the simulation did not run as I expected. 

Would you mind helping me debugging this issue? I would be very grateful for your generosity.

By the way, I am using Unity3D version 2019.4.18f1, and LGSVL 2021.01. All the assets I cloned from GitHub were the latest released version.

Many thanks.

EricBoiseLGSVL, 2021-04-09:
Hello, thanks for trying the latest release of svl simulator.  Simulator work flow has changed quite a bit.  To start simulations in the editor:

1) Be sure to sign up at svlsimulator.com first.  This makes it easy to link the editor to the webUI.
2) Open Loader.scene in Assets/Scenes/ and run the editor
3) You should see a link to cloud button, click it
4) This will open the web browser and ask to login. Do so.  If you are already logged in it will ask to link the machine to the cluster.  Add a name for the machine, I always choose Local.
5) If you haven't logged in before opening the webUI, I recommend stopping Editor after logging in and then playing again to be at cluster naming step.
6) Create simulation and add vehicles, maps, sensors etc and start sim.

Please see documentation [here](https://www.svlsimulator.com/docs/getting-started/getting-started/). The work flow is the same but for editor be sure to start in Loader scene.  All assets are loaded from webUI and attached to vehicle, no need to do it in scene.

wobushigaoying, 2021-04-09:
Hello, thanks for your remainder. I will try your tips soon. 

Hope the simulation will correctly run in Unity3D.

BTW, I noticed "editor" in your reply and some other issues. Does the "editor" you mentioned indicate "Unity3D"?  

Is it convenient to send me projects related to the videos "https://www.youtube.com/watch?v=R_Z65kimCII" and "https://www.youtube.com/watch?v=S0w2ahhEPbE"?

Again, many thanks.

EricBoiseLGSVL, 2021-04-09:
Yes, Unity Editor is usually how it is referenced. We also call it developer mode in the documentation.  It will run in editor no worries, I use it every day :)  Hope the videos are helpful. Let us know if you get stuck.