lgsvl/simulator #1356: Is there any method to get a bird view image from the simulator?

DeepCollision, 2021-04-10:
Hello team, 
I want to get image from bird view, the image contains 3D groud truth box and other information. Is there any way to do that?
[image]

EricBoiseLGSVL, 2021-04-13:
@DeepCollision You will need to make a new GT sensor to include this data.  Then you need to add a camera to the sensor and then send the data in the visualizer method.  If you look at the color camera sensor settings, you see it excludes the "Sensor" layer but the simulator camera includes it.  You can see in WireframeBoxes.cs, the GT box is drawn only to the Sensor layer.  Move the camera in the sensor to a high position and then enable the visualization.  The GT data will then be visible in the camera view.

lemketron, 2021-06-03:
@ssbse2021 Were you able to create this sensor? 

If so, it would be great if you could make the plug-in public so others could use it as well... thanks!

See relevant documentation here: https://www.svlsimulator.com/docs/user-interface/web/library/#editing-a-plugin

closed 2021-07-16 by EricBoiseLGSVL