lgsvl/simulator #1358: How to get Stop Line Sensor Callbacks by Python API?

DeepCollision, 2021-04-11:
Hello team,
I checked the [Test Results webUI page](https://www.svlsimulator.com/docs/user-interface/web/test-results/), shown below:
[image]
and I want to get the callbacks by python APIs,  and I followed the instructions of how to get [Comfor Sensor](https://www.svlsimulator.com/docs/simulation-content/sensors-list/#comfort-sensor) Callback by on_custom() API, and I successed to get the Comfort Sensor Callbacks.
[image]
But when I added [Stop Line Sensor]<https://www.svlsimulator.com/docs/simulation-content/sensors-list/#stopline-sensor> and tried to get the StopLineViolation Callbacks, I found there is no data when the ego viloated the red light.
Anyone know what's the problem? Thanks in advance.

EricBoiseLGSVL, 2021-04-13:
What do you set the stop line sensor threshold in the sensor parameters?  You may need to increase.  Do you have any errors in the playerlog?

DeepCollision, 2021-04-14:
> What do you set the stop line sensor threshold in the sensor parameters? You may need to increase. Do you have any errors in the playerlog?
Hi @EricBoiseLGSVL , I set the threshold to 1, and the sensor setting can be seen in the following:
[image]
 
I collected the Stop Line Callbacks following the steps below:

1. Added a Stop line Sensor to the vehicle using the WebUI configuration page, see following picture:
[image]
2. Created a simulation in API Only mode, see following picture:
[image]
3. Wrote the Callback function for Stop Line Sensor,
[image]
4. Then I called the function on_custom() to caputure the Callbacks:
[image]
5. I ran the script and control the vehicle to violate red light, but there was no message printed. I visualized the sensor message in the simulator and the violation is indeed true.
[image]
6. Then I checked the WebUI to see the test results, all the stop line violations were recorded.
[image]

- But it confused me a lot why there is messge output in my python script. 
- I followed your suggestion to increase the stop line threshold to 10, but the result remained the same as before.
- You suggested to check the playerlog, do you mean i need to check the WebUI of LGSVL?

EricBoiseLGSVL, 2021-04-14:
Ah, I see, sensor is working but the api custom callback is not working.  The custom callback is called in simulator and sends to api.  It's not called in API.  Why are you editing the python api code?

DeepCollision, 2021-04-16:
> Ah, I see, sensor is working but the api custom callback is not working. The custom callback is called in simulator and sends to api. It's not called in API. Why are you editing the python api code?
Hi @EricBoiseLGSVL, I followed the instruction of [API EXAMPLE](https://github.com/lgsvl/ComfortSensor) in Comfort Sensor to write the api code. See below:
[image]

To answer your question about why am I editing the python api code, i just want to get the callback of Stop Line Sensor in my python scripts. Is there any method to get the callbacks by calling apis in python scripts?

EricBoiseLGSVL, 2021-04-16:
Hmm, that looks right and should work.  We'll need to look into this more.  Thanks for clarifying.

DeepCollision, 2021-04-18:
> Hmm, that looks right and should work. We'll need to look into this more. Thanks for clarifying.

Thanks, hope for your reply

ssbse2021, 2021-07-05:
> > Hmm, that looks right and should work. We'll need to look into this more. Thanks for clarifying.
> 
> Thanks, hope for your reply

@EricBoiseLGSVL Hi, is there any method to get the stopline callback by python API? any progross of that?

EricBoiseLGSVL, 2021-07-06:
Sorry, we are trying to get the next release out the door.  We will look into this @revati-naik can you log a bug please?

revati-naik, 2021-07-08:
@ssbse2021 The bug has been reported. Will follow up here if we have any findings. Thanks.

gorkemkrdmn, 2021-08-30:
Would it be possible to use Bridge to access Stop Line Sensor? For example, maybe one can implement a sensor plugin and inside that plugin, it can subscribe to a topic of Bridge.

EricBoiseLGSVL, 2021-08-31:
Yes, that is the best way.  We haven't released source code for stopline sensor, I'll check if we plan to.  In the meantime, yes, create the sensor with the logic and be sure to include the conversion data types/factory type for the bridge.  See DifferentialDriveControlSensor.

KarByc, 2021-10-29:
Hello @ssbse2021 

We've just fixed issue with the StopLineSensor not calling custom callbacks.
It will be available in the next release.
The kind value for the StopLineSensor in the PythonAPI is "stopLineSensor".

Thanks for your detailed report, it was really useful!

closed 2021-10-30 by EricBoiseLGSVL

JasonBourne1998, 2022-06-19:
> Hello @ssbse2021
> 
> We've just fixed the issue with the StopLineSensor not calling custom callbacks. It will be available in the next release. The kind value for the StopLineSensor in the PythonAPI is "stopLineSensor".
> 
> Thanks for your detailed report, it was really useful!

Since I konw there would be no more update right now, could you please give me some advices then I can try to fix it by myself?

KarByc, 2022-06-19:
> Since I konw there would be no more update right now, could you please give me some advices then I can try to fix it by myself?

This issue should be already fixed. AFAIK all the updates were released before closing the project. Are you not getting the custom callbacks on the latest version?

JasonBourne1998, 2022-06-19:
> This issue should be already fixed. AFAIK all the updates were released before closing the project. Are you not getting the custom callbacks on the latest version?

I guess there is no callbacks since I only get comfort sensor
![Uploading 2022-06-19 21-22-30 的屏幕截图.png…]
[image]
()
.
[image]