lgsvl/simulator #1359: Can't create a writer with empty channel name!

DeepCollision, 2021-04-11:
I am running Apollo following these steps:
./docker/scripts/dev_start.sh
./docker/scripts/dev_into.sh
bootstrap.sh
When i want to enable the cyber bridge by typing:
bridge.sh
there came an error saying 'E0411 21:16:49.980556 8788 node_channel_impl.h:154] [cyber_bridge] Can't create a writer with empty channel name!'
[image]

Anyone know what's the problem?

DeepCollision, 2021-04-11:
I got the answer. The sensor deployed on the vehicle need to have a channel name! I will close the issue.

closed 2021-04-11 by DeepCollision

enginlien, 2021-07-29:
Hi DeepCollision

I also encountered the same problem,
can you teach me,
how can i fix this problem.

thanks

[image]

lemketron, 2021-07-29:
Channel names are configured in the sensor json file (see `Topic`), as documented here:
https://www.svlsimulator.com/docs/simulation-content/sensors-list/#how-to-specify-a-sensor