lgsvl/simulator #1360: Issue importing Blender .fbx into SVL simulator

sisaha9, 2021-04-12:
Follow up from #1337

I created a small road network for a track we were planning to use and tried importing into SVL. The textures come out all blackened even after setting materials to use external and take names from model's materials.

Here is how it's appearing on the SVL software

[image]

Attached below is a link to the .fbx file. It has been exported from Blender with experimental transform and textures embedded (done by enabling Copy and Embed during export). I have provided the .blend too if it helps in debugging. Thanks!

https://drive.google.com/drive/folders/1jk-k7Qfmz6B3HZ4Y6fOOl3eO3uM8cwIN?usp=sharing

EricBoiseLGSVL, 2021-04-13:
Do you have the editor light feature on?

[image]

sisaha9, 2021-04-13:
Ah that did reduce the effect a bit but the textures still don't load

[image]

EricBoiseLGSVL, 2021-04-13:
Are the textures embedded in the fbx?  If so, you will need to extract from the model.
[image]

If not you will need to import and create materials to apply the textures.  Then you need to apply the textures to the mesh.
[image]
[image]

sisaha9, 2021-04-13:
I think the issue was my materials were textures. I created new materials with the HDRP pipeline and embedded my textures into it. After that updated all the objects. My grass plane was still not loading. The Unity HDRP template came with a ground object so I modified it to use my material. It is now working

[image]

This screenshot was taken from the default HDRP project but the textures load in the simulator project as well. However, while following the video here: https://www.svlsimulator.com/docs/simulation-content/add-new-map/. I am stuck on the Play button. It shows no cameras rendering and stays

[image]

Is there anything I am doing wrong?

EricBoiseLGSVL, 2021-04-13:
You need to start with Loader scene.  You create the asset bundle with your new map and add it to the simulation you create in WebUI.  Then you can run the map in editor.

sisaha9, 2021-04-13:
Is there a video showing how to start with Loader scene? The video I linked above just created a new scene, added the prefab and running worked

EricBoiseLGSVL, 2021-04-13:
This was made with an older version of simulator.  With the new WebUI you need to run from Loader.scene in Editor.  Just search Loader or go to Assets/Scenes folder.  We have development settings to run from scene but we need more work to support cloud webui.

sisaha9, 2021-04-15:
Is there updated documentation I can follow for building a map then? Or are all the steps in the forementioned video and just for loading I need to take the prefab elsewhere

EricBoiseLGSVL, 2021-04-15:
Documentation is [here](https://www.svlsimulator.com/docs/user-interface/web/library/) for uploading maps.  You need to create asset bundles of the map and upload to webui and add to a simulation to run. Run loader scene, open WebUI, upload map, make simulation with the map, run simulation.

sisaha9, 2021-04-15:
Sorry I should have added. I wanted ways to test during development. I was able to build and successfully run on simulation but everytime I wanted to make a change I would have to go back to the scene, republish, re-add and try which was getting cumbersome

lemketron, 2021-04-15:
You might want to check out the "[Developer Mode](https://www.svlsimulator.com/docs/running-simulations/developer-mode/)" and "[Testing Custom Assets with the Unity editor](https://www.svlsimulator.com/docs/installation-guide/build-instructions/#testing-custom-assets-with-the-unity-editor)" documentation.

sisaha9, 2021-04-15:
I see. I was able to successfully convert the .fbx into a map so that's the end of this issue. Thanks for the resources and all the help in this thread!

closed 2021-04-15 by sisaha9