lgsvl/simulator #1361: How to fix the seed inside the code?

MartinaB85, 2021-04-13:
Good morning,

I'm using your simulator through PythonAPI and I would need to fix the seed inside the code (i.e. in the NPCLaneFollowController.cs script or any other related script) and not through the http://localhost:8080/.

How can I do that?

Thank you in advance for your help,

Martina

EricBoiseLGSVL, 2021-04-13:
@MartinaB85 Hello, you can hardcode this value in Loader.cs on line 125.

MartinaB85, 2021-04-16:
How am I supposed to code it?
The seed is defined inside other functions..should I put simulation.seed = 2020 for example before return in line 125?
Should I change it instead at line 858 ?

Thank you very much,

Martina

lemketron, 2021-04-16:
> I'm using your simulator through PythonAPI and I would need to fix the seed inside the code (i.e. in the NPCLaneFollowController.cs script or any other related script) and not through the http://localhost:8080/.

@MartinaB85 If you're using Python API, you can set the seed in `sim.load` as documented here: https://www.svlsimulator.com/docs/python-api/python-api/#simulation

EricBoiseLGSVL, 2021-04-16:
@lemketron yeah I thought that @MartinaB85 didn't want to use the api but yes it is there

closed 2021-07-16 by EricBoiseLGSVL