lgsvl/simulator #1362: How to add different type of road lanes?  e.g parking lane, Bus lane, Bicycle lane etc

Addi2020, 2021-04-13:
I want to make map visualization by LGSVL-Unity platform instead of using dream view, for which i need to define different type of roads while doing map annotation. base_map.bin is used as a background  and input for building visualization map. 

I need to know how can we can we add more definitions to the road so that we can make good visualization.

The picture below is an example of visualization map that includes parking lanes with purple, bus lane with red etc. 
Picture#1
[image]

whereas, when i upload base_map.bin file to it shows only boundary lines and stop line. 
Picture#2

[image]

EricBoiseLGSVL, 2021-04-13:
@Addi2020 This is not a small undertaking.  You will need to add functionality to mapdata in simulator, in exporter and in map mesh generation.  I'm a little confused on what you are actually trying to do.  You want to replace dreamview with svl simulator?  Can you break this down a little bit more so I can be more clear on what to do?

Addi2020, 2021-04-14:
@EricBoiseLGSVL Thanks for response. I am also not much familiar with this process but, 
Yes, I am planing to use Unity simulator to make my own map visualization. 

Actually, i want to create map visualization similar like the ZOOX show in there video. you can see different types of roads clearly. 

[image]
https://www.youtube.com/watch?v=xgXH4y0s5Ho

So is there anyway to add components (parking lane, color etc ) to maplane while doing map annotation, so that it can be use in creating map visualization.
[image]

EricBoiseLGSVL, 2021-04-14:
Sure you can add any component to every gameobject in scene run time or in editor.  I recommend to modify the MapLane class and add the data you want.  You can do this in editor with public vars first and then edit the map annotation tool later.  Then you will need to alter MapManager.cs or create a sensor plugin to create the overlay objects in scene (look at laneline sensor as an example).  The biggest issue I see is that roads in scene are one mesh and use our environmental effects shader to provide effects to the road.  You may need to alter this or create simple meshes above the road but be careful of z fighting.  Look into Unity tutorials on how to create meshes at runtime and how materials/shaders/colors are applied.  If you just want to use the HDMapMesh for the environment, I recommend modifying this to create the lanes/data/etc you want so it is easy to pull the maplane data and alter the materials.

Addi2020, 2021-04-17:
@EricBoiseLGSVL  I am a little confuse about what you said. could you recommend me web link or video tutorial that i can learn from. the tutorial that can explain things step by step with pictures and examples, it would be better for understanding.

EricBoiseLGSVL, 2021-04-19:
@Addi2020 you will need to look into Unity tutorials.  There you will have multiple videos on how to "edit classes", "add components", "runtime mesh", "material change color", "z fighting", etc.  Search Unity and then the "examples" I posted here.  Start small and work your way to mesh stuff.  You can start with adding the data to map lane and then debug.log at runtime.  Then try to change color on reading that data.

closed 2021-07-16 by EricBoiseLGSVL