lgsvl/simulator #1363: What are the information details of the topic （/vehicle_cmd）？

cxf99, 2021-04-13:
Is it - VehicleControlData.msg  ？

@lemketron

cxf99, 2021-04-16:
Excuse me, can anyone answer that？

EricBoiseLGSVL, 2021-04-23:
Sorry, what do you mean?  Can you explain your issue a little more?

This is the class that is in the source code.
```
public enum GearPosition
    {
        Neutral,
        Drive,
        Reverse,
        Parking,
        Low,
    };

    public class VehicleControlData
    {
        // common
        public float? Acceleration; // 0..1
        public float? Braking; // 0..1

        // autoware
        public float? Velocity;
        public float? SteerAngularVelocity;
        public float? SteerAngle;
        public bool ShiftGearUp;
        public bool ShiftGearDown;

        // apollo 
        public float? SteerRate;
        public float? SteerTarget;
        public double? TimeStampSec;
        public GearPosition? CurrentGear;

        // lgsvl
        public float? SteerInput;
    }
```