lgsvl/simulator #1364: Asset Bundle Upload Error

ijb5117, 2021-04-14:
I'm trying to have cars of different colors for simulation purposes. I used the Jaguar2015XE as a model and changed its color on Unity. After building the Asset Bundle, (size of 4,193KB) I am trying to upload it in my LGSVL library, but I get the following messages:
"Unexpected error occurred. Please try again later or contact administrator"
"Server Error - check the terminal for more info!"

lemketron, 2021-04-23:
We are investigating problems with bundle uploads occasionally failing. Please see if you are able to unzip the bundle and if not, try building again. It seems that sometimes the bundle build fails but the file is still generated and it's not clear that it is invalid until you try to upload it. We should add a more helpful error message on upload and also figure out why the build is sometimes failing. (tracking in AUTO-6845)

GeraldGuo7, 2021-04-28:
I also have the same issue even if I tried to download the lgsvl map assetbundle from the lgsvl official site and upload it. I make sure that the bundle is built correctly and I can also unzip it. So I want to know how to fix this upload error? Is this a network problem?

lemketron, 2021-04-28:
> I also have the same issue even if I tried to download the lgsvl map assetbundle from the lgsvl official site and upload it. I make sure that the bundle is built correctly and I can also unzip it. So I want to know how to fix this upload error? Is this a network problem?

@GeraldGuo7 Can you describe in more detail exactly what you're trying to do? What map did you download and from where, what version of simulator are you using to build it, and can you share a link to the resulting zip file? Thanks!

GeraldGuo7, 2021-05-10:
Hi!lemketron,
Sorry about late reply because of a 5-day Chinese holiday and the dedicated work about other things so I spent some time to siwtch to the LGSVL after I was stuck in some unknown errors which looks like a network problem.
The prerequistes:My LGSVL version is windows64 2021.1 and what I actually want to do is to use the logitechg29 steering wheel to control my vehicle in simulation. After some struggles in the internet , I finally decided to construct my own g29 driver in one example map 'cubetown',  
[image]

and I tried to upload new map（actually just some configurations for logitechg29 were changed,I couldn't upload my environment_CubeTown asset here because of unsupported type)
 to the maps library in my LGSVL account and I always failed ,here are some error results,
[image]

Another issue is to avoid the issues other than the map asset, I downloaded the 'Shalun' map from https://content.lgsvlsimulator.com/maps/shalun/, and I got the environment_Shalun file, and after I extracted it, I got these 4 files:manifest, 6e44b039-36a2-4541-9396-615b48258393_environment_textures, 6e44b039-36a2-4541-9396-615b48258393_environment_main_windows, 6e44b039-36a2-4541-9396-615b48258393_environment_main_linux.
It looks pretty basic and I click the add new map in my maps library and there still exists some error,
[image]

Thanks a lot for any information!

eagafonovlge, 2021-05-27:
@GeraldGuo7 

>  I downloaded the 'Shalun' map from https://content.lgsvlsimulator.com/maps/shalun/

That assets are not compatible with Simulator 2021.1. The page states
> Supported LGSVL Simulator versions: 2020.06

Simulator 2021.1 expects mainfest in JSON format, so you should see `manifest.json` file in the newly build asset bundle.

GeraldGuo7, 2021-06-10:
@eagafonovlge 
Thanks for your accurate correction. I was struggling against how to build asset bundle, create my own map or modify the original lgsvl map to meet some specific requirements. 
Currently, I put aside it but in the near future, I will pick it up.Hope there will be more uptodate lgsvl tutorials.
Anyway, thanks for your reply.

kk2491, 2021-06-25:
@GeraldGuo7 I am also facing similar issue. 
I am trying to upload a new vehicle from here : https://content.lgsvlsimulator.com/vehicles/. 
When I try to upload I get the below error:  

[image]
Simulator Version : master  
Unity Editor : 2020.3.3f1  

Kindly let me know how can I upload.

Thank you

lemketron, 2021-06-25:
The AWF Lexus vehicle is already available for SVL Simulator here:
https://wise.svlsimulator.com/vehicles/profile/b22952f6-f82b-4eeb-ac28-2067e88569a1

The assets you download from the content.lgsvlsimulator.com site are only for use with LGSVL Simulator 2020.06 and will not work with the wise.svlsimulator.com site nor with the 2021.x versions of SVL Simulator.

All previously available LGSVL Simulator 2020 vehicles have been updated for SVL Simulator 2021 and can be found here:
https://wise.svlsimulator.com/vehicles/public

lemketron, 2021-06-25:
If you are looking for the source to build a vehicle for SVL Simulator 2021 then you probably want to check out the `2021.1` tag of the `Jaguar2015XE` repository here: https://github.com/lgsvl/Jaguar2015XE/tree/2021.1

Make sure you are using the 2021.1.1 tag of the SVL Simulator project as well and NOT master (which is newer, and not compatible) to build the vehicle asset bundle.

kk2491, 2021-06-25:
Thanks for explaining the problem. 
I was trying to create my own vehicle and trying to upload the asset to the wise.svlsimulator.com. 
Just to give a try I thought to download the freely available assets and upload it to wise.svlsimulator.com. As you answered, it showed me unexpected error. 

As you suggested, as a first step I will move to https://github.com/lgsvl/simulator/tree/2021.1.1. And try to generate asset bundle by following the https://github.com/lgsvl/Jaguar2015XE/tree/2021.1 and upload it to the wise.svlsimulator.com instead of downloading the available ones. 

I will come back if there are any queries. 

Thank you, appreciate your help.

danielm1405, 2021-06-28:
I have the same issue while trying to upload a sensor plugin. I built https://github.com/lgsvl/VehicleStateSensor (to learn how to add custom sensor and develop it in the future). The build seems to be correct, there were no errors. However, the upload resulted in the following error:

[image]

My configuration:
* OS: Ubuntu 20.04
* SVL version: 2021.1.1
* Unity version: 2019.4.18f1
* VehicleStateSensor at [this](https://github.com/lgsvl/VehicleStateSensor/commit/fd155fcb5bf9c29c70703cb9d90356c235e17d02) commit (as newest version from `master` resulted in `no suitable method found to override` errors)
* Resulting AssetBundle: [sensor_VehicleStateSensor.zip](https://github.com/lgsvl/simulator/files/6727317/sensor_VehicleStateSensor.zip)

Any help would be appreciated, thank you in advance.

closed 2021-07-16 by EricBoiseLGSVL