lgsvl/simulator #1365: Can't upload custom sensor

cubrink, 2021-04-14:
I'm testing building custom sensors in SVL 2021.1 but I'm getting an error when I upload my sensors to https://wise.svlsimulator.com/plugins/profile/add. 

First, I've successfully built and uploaded the sample custom sensor available [here](https://github.com/lgsvl/ComfortSensor).

My next attempt was to make a modified version of an existing sensor. I decided on the [CanBusSensor](https://github.com/lgsvl/CanBusSensor). Before actually modifying the behavior of the sensor I attempted to build an upload the CanBusSensor under a different name, `CustomCanBusSensor`. 

I was able to successfully build but I get the following error when trying to upload:
[image]

It says to check the terminal, but I'm unsure how to access that information.

I've since deleted both plug-ins and rebuilt both. The error is reproducible. 

To create the `ComfortSensor` plugin, I just `git clone` the repo in `simulator/Assets/External/Sensors/` and then build and upload. There are no errors in the process.

To create the `CustomCanBusSensor` sensor I've followed the details in [this](https://www.svlsimulator.com/docs/plugins/sensor-plugins/) guide.

As an overview, I do the following:
1. Create a new folder in `simulator/Assets/External/Sensors/` called `CustomCanBusSensor`
2. Place a new prefab with the same name in the folder
3. On the prefab, Add Component -> SensorsController
4. Inside the folder, I place a copy of the `CanBusSensor.cs` from the [CanBusSensor](https://github.com/lgsvl/CanBusSensor) repo
5. Then I make the following modifications:
    - Change the filename to `CustomCanBusSensor.cs`
    - Change `public partial class CanBusSensor : SensorBase` to `public partial class CustomCanBusSensor : SensorBase`
 
6. I build the sensor using in Unity using Simulator -> Build. I checking `CustomCanBusSensor` listed under sensors, then click 'Build'. 

The sensor builds successfully, but I get the error when I try to upload. Is there something I've forgotten to change or a file that I haven't copied?

System specs: 
OS: Windows 10 64-bit, Build 20H2
Unity: Version 2019.4.18f1

MisterioRemo, 2021-04-14:
@cubrink hello.

> 3. On the prefab, Add Component -> SensorsController

I think it should be `Add Component -> CustomCanBusSensor.cs`

cubrink, 2021-04-14:
@MisterioRemo Yep, that fixed it. I must been misreading the guide. Thanks.

closed 2021-04-14 by cubrink