lgsvl/simulator #1366: Some questions about velodyne lidar

qiongwu1003, 2021-04-15:
Hi, SVL team

About velodyne could I know following informations？

1. In the actual implementation, what information does this lidar model consume as input from the 3D model to generate simulated pointcloud data?
2. For different sensor types (Velodyne 16 vs Velodyne 128), does the Lidar model require more data from the 3D engine to provide dense pointclouds? 
3. What is the communication interface between 3D Model and LIDAR sensor model？ UDP, right？

Thanks.

EricBoiseLGSVL, 2021-04-16:
1) Depth, intensity and position from render texture, not mesh
2) yes, the greater the density of rays, the greater the data
3) yep, UDP
No problem

qiongwu1003, 2021-04-26:
@EricBoiseLGSVL Thanks for your reply.

closed 2021-04-26 by qiongwu1003