lgsvl/simulator #1367: Apollo Planning Error

EricWu416, 2021-04-15:
I'm trying to run the simulation with Apollo 6.0 and followed this [process](https://www.svlsimulator.com/docs/system-under-test/apollo-master-instructions/). I was able to connect LGSVL and Apollo since I can see the CyberRT messages from the simulation in `cyber_monitor`. However, when I send the routing request, the planning module complains "Failed to update reference line after rerouting" consistently. The planning module would work just fine in sim control instead. Any suggestions or recommendations to resolve this issue will be really helpful. Thanks!

EricBoiseLGSVL, 2021-04-16:
@lemketron ?

salmanee, 2021-04-19:
Any update son this issue @lemketron @EricBoiseLGSVL

lemketron, 2021-04-23:
@salmanee Can you explain how you are sending the routing request and what you mean by it works fine in sim control instead?

Also, I have found that if Apollo has already driven to a destination, the planning module usually needs to be restarted (disabled and re-enabled) before trying to run the same scenario again.

EricWu416, 2021-04-24:
@lemketron I used the routing editing to set two waypoints in dreamview. After sending the routing request I can see the errors from planning. The error only occurs when running Apollo with LGSVL. I followed the same steps (selecting waypoints and sending the routing request) with dreamview and planning was able to produce trajectories without errors. Hope this can answer your questions.

irfanbadshah, 2021-04-30:
Hi, 

I have a simillar  issue but when i run it with LGSVL simulator. 

**System information**
**OS Platform and Distribution (e.g., Linux Ubuntu 18.04)**: Linux Ubuntu 18.04
**Apollo version (3.5, 5.0, 5.5, 6.0)**: 5.0, 6.0 [LGSVL fork]
**LGSVL**: Windows 

My issue is that when I turn on the planning module in the Drieamview and send a routing request, the planning module gets disabled and the Module Delay for the planning module on the Dreamview keeps increasing. I also get a "Planning trggers safe mode message" in the Console window.
[image]

 The "/apollo/control "channel gives the msg "planning has no trajectory point. planning_seq_num:5"

[image]

Here is a list of channels thats are sent from LGSVL to the Apollo.
[image]

Thanks